dreemurrs-embedded/Pine64-Arch #515: Pinephone cpuidle power saving broken

emulti, 2023-01-13:
- Device: Pinephone 1.2b
- Kernel Version : linux-megi 6.0.10-1-danctnix
- UI: plasma-mobile

### Steps to reproduce
Boot phone, update to latest packages, install cpupower utility, check for items below

### Expected behavior
(note: taken from https://gitlab.com/postmarketOS/pmaports/-/issues/1901
```
pine64-pinephone:~# cpupower idle-info
CPUidle driver: psci_idle
CPUidle governor: teo
analyzing CPU 0:

Number of idle states: 3
Available idle states: WFI cpu-sleep cluster-sleep
WFI:
Flags/Description: ARM WFI
Latency: 1
Usage: 138252
Duration: 330496749
cpu-sleep:
Flags/Description: cpu-sleep
Latency: 2300
Usage: 457
Duration: 6620832
cluster-sleep:
Flags/Description: cluster-sleep
Latency: 2350
Usage: 91
Duration: 2605765
```

### Actual behavior
on  linux-megi 6.0.10-1-danctnix image:
```
pine64-pinephone:~$ cpupower idle-info
CPUidle driver: none
CPUidle governor: menu
analyzing CPU 0:

CPU 0: No idle states
```

### Logfiles and additional information
See https://gitlab.com/postmarketOS/pmaports/-/issues/1901
Seen on latest released danctnix image and also same image updated to latest kernel 6.0.1
Issue appears to be caused by overwriting DTB in u-boot user script, so the cpuidle nodes added by ARM TF-A are not used.
However, at time of this posting, fixing the overwrite causes major regressions such as loss of charging, screen brightness etc, which are being worked on.
Megi's p-boot does not exhibit this issues as it passes the TF-A modified DTB direct to the kernel. I intend to try and test this to confirm

Danct12, 2023-01-13:
Hey, thanks for reporting this one. This is a very serious bug.

emulti, 2023-01-13:
Yes, I noticed it on PostmarketOS and the Danctnix distros when investigating why the battery runtime was so short...

I think it could affect other distributions too due to the tendency to share base-level work like u-boot.

I tried to test with p-boot, but have reached a block at the moment because your images use a GPT partition table and p-boot only supports MSDOS/FAT.

emulti, 2023-01-17:
Now fixed in PostmarketOS master branch with change to u-boot script

https://gitlab.com/postmarketOS/pmaports/-/merge_requests/3806

marek-lach, 2023-01-17:
@Danct12 Megi wrote a blog post about implementing this in distros: https://xnux.eu/log/#077

Danct12, 2023-01-19:
New kernel and U-Boot available in testing, please check it out to see if this fixes the issue for you. It does for me:

```
[alarm@danctnix ~]$ sudo cpupower idle-info
[sudo] password for alarm: 
CPUidle driver: psci_idle
CPUidle governor: menu
analyzing CPU 0:

Number of idle states: 3
Available idle states: WFI cpu-sleep cluster-sleep
WFI:
Flags/Description: ARM WFI
Latency: 1
Usage: 34591
Duration: 27730553
cpu-sleep:
Flags/Description: cpu-sleep
Latency: 2300
Usage: 1020
Duration: 3543460
cluster-sleep:
Flags/Description: cluster-sleep
Latency: 2350
Usage: 354
Duration: 846149
```

emulti, 2023-01-19:
No apparent regressions for me. cpuidle working again.

closed 2023-01-25 by Danct12

pentamassiv, 2023-01-28:
When I run the command, I get: 

```
[alarm@pinephone ~]$ sudo cpupower idle-info
CPUidle driver: psci_idle
CPUidle governor: menu
analyzing CPU 0:

Number of idle states: 3
Available idle states: WFI cpu-sleep cluster-sleep
WFI:
Flags/Description: ARM WFI
Latency: 1
Usage: 1583163
Duration: 3619225080
cpu-sleep:
Flags/Description: cpu-sleep
Latency: 2300
Usage: 0
Duration: 0
cluster-sleep:
Flags/Description: cluster-sleep
Latency: 2350
Usage: 0
Duration: 0
```

Is it expected to see:
```
Usage: 0
Duration: 0
```
?