dreemurrs-embedded/Pine64-Arch #517: New uboot doesn't know btrfs

DrRac27, 2023-01-25:
- Device: PinePhone
- Kernel Version : linux-megi-6.1.7-1-aarch64.pkg.tar.xz, can't $(uname -r)
- UI: Phosh but doesn't start so doesn't matter

### Steps to reproduce

I have been using the same install of arch for a long time. I used FDE + btrfs to install it and until now everything worked but now i compiled the new kernel and the new u-boot (uboot-pinephone-2023.01-1-aarch64.pkg.tar.xz), made a full update and now it boots into an emergency shell with the error message: `mount: /new_root: unknown filesystem type 'btrfs'.`
Did you disable btrfs in the u-boot config?

### Expected behavior

Boot as every time before.

### Actual behavior

Boots into emergency shell. Doesn't open a osk so I can't type and find out more because I have a unpatched UBPorts Edition where the USB data pins have to be soldered somehow (didn't look into it yet).

### Logfiles and additional information

FDE unlocking works.

Danct12, 2023-01-25:
The btrfs module is not built in, but rather as a module.

Try boot to another Linux install, mount root to /mnt and boot to /mnt/boot. Chroot in, add 'btrfs' into list of modules in /etc/mkinitcpio.conf and regenerate initramfs.

DrRac27, 2023-01-25:
This is nothing that changed so I didn't think it could cause the problem. And in fact all that was necessary was to regenerate the initramfs. Maybe my `-Syu` failed? It is working again.
I am a bit confused because I checked my `/etc/mkinitcpio.conf` and the modules array is empty and it works anyways?

closed 2023-01-25 by DrRac27