dreemurrs-embedded/Pine64-Arch #523: Revert "usb: quirks: Add USB_QUIRK_RESET for Quectel EG25G Modem"

Shatur, 2023-02-07:
- Device: PinePhone Pro
- Kernel Version : 6.1.7
- UI: Phosh

### Steps to reproduce

Run `sudo systemctl suspend` and wake up the phone by pressing the power button.

### Expected behavior

Modem is available.

### Actual behavior

Modem reappears in ~10 seconds.

### Other distros

The issue was fixed in other distros such as Mobian by [reverting the quirk](https://gitlab.com/marcinipv8/kernel-ark/-/commit/fedafc567167a7a0d10ddc5657dd3f32a872ec86). I think we should do it as well.

### Temporary workaround without recompiling the kernel

Just add `usbcore.quirks=2c7c:0125:e` to the kernel boot parameters in `boot/boot.txt` and run `sudo ./mksrc`.

Shatur, 2023-02-10:
Even though the modem stops to disappear, a message from qmi about incoming call from suspend may be lost upon waking up.
So there is additional patch that is required for PPP: [usb-serial-option-add-reset-resume-callback-for-WWAN](https://salsa.debian.org/Mobian-team/devices/kernels/rockchip-linux/-/blob/mobian-6.1/debian/patches/common/0006-usb-serial-option-add-reset-resume-callback-for-WWAN.patch).
I compiled the kernel with this patch and can confirm that it fixes the issue. Please, add this patch too. Was suggested by `@biktorgj`.

bobjrsenior, 2023-02-17:
Looks like the requested changes were added in this commit: https://github.com/dreemurrs-embedded/Pine64-Arch/commit/00247b8918753c4a89dfab50f656edb63f927984

I installed the latest changes and everything seems to be working so far. The modem is appearing right away when waking from suspend.

Shatur, 2023-03-02:
Was fixed, the problem no longer appears.

closed 2023-03-02 by Shatur