dreemurrs-embedded/Pine64-Arch #536: Auto Rotation is enabled but device is not switching to landscape

bmosley, 2023-03-25:
- Device: PinePhone
- Kernel Version : 6.1.12-1-danctnix
- UI: default

### Steps to reproduce
Newly installed image from the latest release.
$ sudo pacman -Syu
Auto Rotation is enabled but device is not switching to landscape.

### Expected behavior
Rotation of the device (while auto orientation is enabled) should change orientation of the UI

### Actual behavior
Stuck in portrait orientation

### Logfiles and additional information
[journalctl.txt](https://github.com/dreemurrs-embedded/Pine64-Arch/files/11067554/journalctl.txt)

UltraBlackLinux, 2023-03-26:
Same. Quoting what I wrote into phosh's matrix room:

> does anyone else have issues with the power button since the update that added these gnome 43 like buttons into the drop down menu? Pressing the power button like normal instantly wakes up the phone again. I have to really tap it as short as possible to turn off the screen and still it does not lock my phone

> also automatic display rotation is really iffy since then. It only works like 20% of the time

Jeremiah-Roise, 2023-04-13:
#409
For anybody who wants a temporary fix until iio-sensor-proxy v3.5 is released just delay `iio-sensor-proxy` by adding `ExecStartPre= /bin/sleep 1` to
`/usr/lib/systemd/system/iio-sensor-proxy.service`  on the line above `ExecStart=`

Danct12, 2023-04-15:
Duplicate of #409.

closed 2023-04-15 by Danct12