dreemurrs-embedded/Pine64-Arch #537: Install wireplumber and gst-plugin-pipewire by default (to enable Pipewire cameras)

rmader, 2023-03-26:
- Device: PinePhonePro
- Kernel Version : 6.1.12
- UI: Phosh

The future/modern camera stack on Linux, consisting of libcamera and pipewire (similar to alsa + pulseaudio/pipewire) provides apps with a clean API to access cameras without needing device specific knowledge.

As of today (Mar. 26) with the libcamera 0.0.4-3 upgrade this almost works on the Pinephone Pro on a vanilla installation of the archlinux-pinephone-pro-phosh-20230203.img.xz image.

The only changes needed to make applications like Cheese be able to use the cameras now are:
1. install `wireplumber` (needed to make Pipewire actually expose cameras - requires a restart)
2. install `gst-plugin-pipewire` (implies 1.)
3. the pending kernel update to 6.2.7

These packages get installed by default on most distros nowadays - it would be nice if the image hear would do so as well.

Danct12, 2023-03-26:
Kernel 6.2.7 is in the testing repository for days now, I'll move it to main later as there's not yet a problem reported on the kernel.

rmader, 2023-03-26:
> Kernel 6.2.7 is in the testing repository for days now, I'll move it to main later as there's not yet a problem reported on the kernel.

Thanks, just updated and confirmed again that it works.

UltraBlackLinux, 2023-03-27:
@grinceur bruh which discord server -_-

rmader, 2023-03-28:
> megapixel is broken for me with this kernel…

Note: this discussion here is not about Megapixels. It's about the proper libcamera + pipewire (+ gstreamer + xdg-camera-portal) stack, which AFAIK Megapixels currently does not plan to support.

Apps that use this stack include Cheese, [the upcoming, mobile friendly, new Gnome camera app](https://gitlab.gnome.org/msandova/snapshot) and soon also [Chromium](https://chromium-review.googlesource.com/c/chromium/src/+/3308882), Firefox and OBS-Studio.