dreemurrs-embedded/Pine64-Arch #541: PineTab2 kernel fails to compile

oklopfer, 2023-04-23:
- Device: PineTab2
- Kernel Version : 6.2.0 - 6.3.0-rc7

### Steps to reproduce
Attempt to compile the kernel

### Expected behavior
Kernel compiles successfully

### Actual behavior
Kernel build fails

### Logfiles and additional information
`oklopfer@oklopfer:~/linux$ ARCH=arm64 CROSS_COMPILE=aarch64-linux-gnu- KBUILD_DEBARCH=arm64 LOCALVERSION=-oktab make -j$(nproc) modules
  CALL    scripts/checksyscalls.sh
  CC [M]  drivers/net/wireless/bes/bes2600/main.o
  CC [M]  drivers/net/wireless/bes/bes2600/bes2600_sdio.o
drivers/net/wireless/bes/bes2600/main.c: In function ‘bes2600_init_wapi_cipher’:
drivers/net/wireless/bes/bes2600/main.c:326:23: error: variable ‘wapi_cs’ has initializer but incomplete type
  326 |         static struct ieee80211_cipher_scheme wapi_cs = {
      |                       ^~~~~~~~~~~~~~~~~~~~~~~
drivers/net/wireless/bes/bes2600/main.c:327:18: error: ‘struct ieee80211_cipher_scheme’ has no member named ‘cipher’
  327 |                 .cipher = WLAN_CIPHER_SUITE_SMS4,
      |                  ^~~~~~
In file included from ./include/net/mac80211.h:20,
                 from drivers/net/wireless/bes/bes2600/main.c:22:
./include/linux/ieee80211.h:3955:25: warning: excess elements in struct initializer
 3955 | #define SUITE(oui, id)  (((oui) << 8) | (id))
      |                         ^
./include/linux/ieee80211.h:3972:41: note: in expansion of macro ‘SUITE’
 3972 | #define WLAN_CIPHER_SUITE_SMS4          SUITE(0x001472, 1)
      |                                         ^~~~~
drivers/net/wireless/bes/bes2600/main.c:327:27: note: in expansion of macro ‘WLAN_CIPHER_SUITE_SMS4’
  327 |                 .cipher = WLAN_CIPHER_SUITE_SMS4,
      |                           ^~~~~~~~~~~~~~~~~~~~~~
./include/linux/ieee80211.h:3955:25: note: (near initialization for ‘wapi_cs’)
 3955 | #define SUITE(oui, id)  (((oui) << 8) | (id))
      |                         ^
./include/linux/ieee80211.h:3972:41: note: in expansion of macro ‘SUITE’
 3972 | #define WLAN_CIPHER_SUITE_SMS4          SUITE(0x001472, 1)
      |                                         ^~~~~
drivers/net/wireless/bes/bes2600/main.c:327:27: note: in expansion of macro ‘WLAN_CIPHER_SUITE_SMS4’
  327 |                 .cipher = WLAN_CIPHER_SUITE_SMS4,
      |                           ^~~~~~~~~~~~~~~~~~~~~~
drivers/net/wireless/bes/bes2600/main.c:328:18: error: ‘struct ieee80211_cipher_scheme’ has no member named ‘iftype’
  328 |                 .iftype = BIT(NL80211_IFTYPE_STATION),
      |                  ^~~~~~
In file included from ./arch/arm64/include/asm/alternative-macros.h:6,
                 from ./arch/arm64/include/asm/barrier.h:14,
                 from ./include/linux/list.h:11,
                 from ./include/linux/module.h:12,
                 from drivers/net/wireless/bes/bes2600/main.c:14:
./include/vdso/bits.h:7:33: warning: excess elements in struct initializer
    7 | #define BIT(nr)                 (UL(1) << (nr))
      |                                 ^
drivers/net/wireless/bes/bes2600/main.c:328:27: note: in expansion of macro ‘BIT’
  328 |                 .iftype = BIT(NL80211_IFTYPE_STATION),
      |                           ^~~
./include/vdso/bits.h:7:33: note: (near initialization for ‘wapi_cs’)
    7 | #define BIT(nr)                 (UL(1) << (nr))
      |                                 ^
drivers/net/wireless/bes/bes2600/main.c:328:27: note: in expansion of macro ‘BIT’
  328 |                 .iftype = BIT(NL80211_IFTYPE_STATION),
      |                           ^~~
drivers/net/wireless/bes/bes2600/main.c:329:18: error: ‘struct ieee80211_cipher_scheme’ has no member named ‘hdr_len’
  329 |                 .hdr_len = 18,
      |                  ^~~~~~~
drivers/net/wireless/bes/bes2600/main.c:329:28: warning: excess elements in struct initializer
  329 |                 .hdr_len = 18,
      |                            ^~
[... 53 lines cut ...]
  326 |         static struct ieee80211_cipher_scheme wapi_cs = {
      |                                               ^~~~~~~
make[6]: *** [scripts/Makefile.build:252: drivers/net/wireless/bes/bes2600/main.o] Error 1
make[6]: *** Waiting for unfinished jobs....
drivers/net/wireless/bes/bes2600/bes2600_sdio.c: In function ‘bes2600_get_gpio_from_dts’:
drivers/net/wireless/bes/bes2600/bes2600_sdio.c:1403:28: error: storage size of ‘flags’ isn’t known
 1403 |         enum of_gpio_flags flags;
      |                            ^~~~~
drivers/net/wireless/bes/bes2600/bes2600_sdio.c:1407:31: error: implicit declaration of function ‘of_get_named_gpio_flags’; did you mean ‘of_get_named_gpio’? [-Werror=implicit-function-declaration]
 1407 |                 wakeup_gpio = of_get_named_gpio_flags(wireless_node, gpio_name, 0, &flags);
      |                               ^~~~~~~~~~~~~~~~~~~~~~~
      |                               of_get_named_gpio
drivers/net/wireless/bes/bes2600/bes2600_sdio.c:1403:28: warning: unused variable ‘flags’ [-Wunused-variable]
 1403 |         enum of_gpio_flags flags;
      |                            ^~~~~
cc1: some warnings being treated as errors
make[6]: *** [scripts/Makefile.build:252: drivers/net/wireless/bes/bes2600/bes2600_sdio.o] Error 1
make[5]: *** [scripts/Makefile.build:494: drivers/net/wireless/bes/bes2600] Error 2
make[4]: *** [scripts/Makefile.build:494: drivers/net/wireless/bes] Error 2
make[3]: *** [scripts/Makefile.build:494: drivers/net/wireless] Error 2
make[2]: *** [scripts/Makefile.build:494: drivers/net] Error 2
make[1]: *** [scripts/Makefile.build:494: drivers] Error 2
make: *** [Makefile:2025: .] Error 2`

I am in charge of porting the PineTab 2 to Ubuntu Touch, and I am failing to build the kernel due to the `bes2600` module being incomplete, and therefore unable to compile. I have looked across your repository and pkgs, but have not been able to figure out how you are getting around this issue.

Danct12, 2023-04-23:
BES2600 module currently only compiles with the set of options enabled in this [kernel config](https://github.com/dreemurrs-embedded/Pine64-Arch/blob/master/PKGBUILDS/pine64/linux-pinetab2/config#L2181-L2206)

I'll be closing this as this isn't related to this distro, please join the [PineTab chat room](https://wiki.pine64.org/wiki/Main_Page#Chat_Platforms) or ask in UBports Porting chat (will catch you on there).

closed 2023-04-23 by Danct12

oklopfer, 2023-04-24:
Got it, thanks. Missed having to actually disable those other modules for it. Compiling successfully on 6.3 now.