dreemurrs-embedded/Pine64-Arch #567: pkg: danctnix: mmsd-tng is out of date

mikeshenson, 2023-07-21:
- Device: PinePhonePro
- Kernel Version : 6.4.1-1-danctnix
- UI: Phosh 0.28
- Chatty: 0.7.3-2

### Steps to reproduce
Open Chatty, There is no "Attachment" icon by the text box.

### Expected behavior
I compile version 2.3.0 of mmsd-tng and installed it.
I rebooted the phone and the "Attachment" icon was back.

### Logfiles and additional information
### mmsd-tng 2.0.0
journalctl
Jul 20 19:23:03 danctnix mmsdtng[5122]: /usr/bin/mmsdtng: symbol lookup error: /usr/bin/mmsdtng: undefined symbol: _ZN4i18n12phonenumbers11PhoneNumberC1EPN6google8protobuf5ArenaEb

### mmsd-tng 2.3.0
[alarm@danctnix ~]$ systemctl --global status mmsd-tng.service
mmsd-tng.service - Multimedia Messaging Service Daemon
     Loaded: loaded (/usr/lib/systemd/user/mmsd-tng.service; enabled; preset: enabled)
     Active: active (running) since Thu 2023-07-20 20:05:24 CDT; 1min 9s ago
   Main PID: 3820 (mmsdtng)
      Tasks: 4 (limit: 4326)
     Memory: 14.3M
        CPU: 236ms
     CGroup: /user.slice/user-1000.slice/user@1000.service/app.slice/mmsd-tng.service
             └─3820 /usr/bin/mmsdtng -d

kclisp, 2023-07-25:
I believe this breaks MMS as well.

Danct12, 2023-07-28:
Upgraded to 2.4.0 in 8b50f4a2e1b1685ff3eb09c41788b267f4abfe0e

closed 2023-07-28 by Danct12