dreemurrs-embedded/Pine64-Arch #570: On Screen Keyboard not working with desktop browsers

dozadoesit, 2023-07-31:
------------------------------------------------------------------------------->

- Device: pinetab2
- Kernel Version : 6.4.3-danctnix1-1-pinetab2
- UI: KDE Plasma

### Steps to reproduce
Open a desktop browser and select a text field the on screen keyboard does not pop up. Try to enter numerals and you get special characters. ( tested in Brave and Firefox )

### Expected behavior
When selecting text fields in desktop browsers the onscreen keyboard should appear. Also the onscreen keyboard should be able to send numerals other wise you won't be able to log in to something if a password has numerals. 

### Actual behavior
The on screen keyboard does not appear. I can manually pull it up but if I type numerals in a text field instead of 123 I get !@# and there is no way to get numeral characters.

dozadoesit, 2023-09-12:
I wanted to add to this that there are times where you select the ^ for the on screen keyboard to manually get it to pop up. Sometimes that work and sometimes no matter how many times I click on it the on screen keyboard won't come up

ReinerText, 2024-08-23:
* Device: pinetab2
* Kernel Version : 6.9.9-danctnix1-1-pinetab2
* UI: KDE Plasma

I can't reproduce this bug with Firefox 129.0.2