dreemurrs-embedded/Pine64-Arch #571: Switch to extlinux.conf

Shatur, 2023-08-01:
Recently megi released [rk2aw with patched U-Boot builds](https://xnux.eu/log/090.html)
This bootloaders have many benefits (especially for PPP), but looks like patched U-Boot uses `extlinux.conf` while `boot.scr`. I would suggest to migrate to simplify setup for users.
I tried creating `extlinux/extlinux.conf` with the following content:
```
MENU TITLE Pinephone Pro Boot Menu
MENU BACKGROUND /splash.bmp

LABEL emmc
        MENU LABEL eMMC - Arch Linux
        FDT /dtbs/rockchip/rk3399-pinephone-pro.dtb
        KERNEL /Image
        INITRD /initramfs-linux.img
        APPEND loglevel=4 console=ttyS2,115200 console=tty0 root=/dev/mmcblk2p2 rw rootwait fbcon=rotate:1 pinephone_keyboard.disable_input
```
But it doesn't work for some reason...

Shatur, 2023-08-02:
> But still unable to boot. First item just boots me into console.

Put extlinux.conf back and now it boots.

Shatur, 2023-08-02:
@Danct12 I think `extlinux.conf` is way better and simpler way configure the system. I would suggest to migrate to it.

Shatur, 2023-08-02:
I talked with danct and looks like we can't because we need dynamically identify a boot device on boot up and it's impossible from `extlinux.conf`.

`boot.scr` doesn't work because we megi's builds missing `CONFIG_CMD_UNZIP` enabled. We could enable it.

closed 2023-08-02 by Shatur

Shatur, 2023-08-03:
I talked with megi and he enabled `CONFIG_CMD_UNZIP`. So now it boots without configuration.

oklopfer, 2023-08-11:
An easy way to provide this in a package is by using the Mobian team's [fork of u-boot-menu](https://salsa.debian.org/Mobian-team/packages/u-boot-menu/).