dreemurrs-embedded/Pine64-Arch #572: Phone run super hot and will barelly charge on last update (regress)

bfra2373, 2023-08-02:
- Device: (PinePhone 1.2)
- Kernel Version : 6.1-sunxi64 on mobian but same issue with arch mobile
- UI: phosh

### Steps to reproduce
Turn the phone on

### Expected behavior
Phone should remain cold to touch and keep it's charge for at least a day

### Actual behavior
Phone run super hot even when not in use. Battery life is now about 1 hours (it will go down about 2-3% every minute) and will not charge unless the phone is plugged to a high amp power supply.

### Logfiles and additional information

bfra2373, 2023-08-03:
A bit more detail.
Before update phone was running at 2watt average
After update  the phone was running at 4Watt average 
Re-updated to latest and now it run at 2.4Watt average (a 20% increase in power consumption compare to older version).
The phone is idle in all case with brightness at 50% and no thing open except Powersupply app

grinceur, 2023-08-11:
can you try to switch off the wifi from the kill switches and see how the power consumption goes ? I had to do this with mine, it was getting super hot too… no issues since i switched it off

bfra2373, 2023-08-11:
tested and well something kinda weird happen... 
while discharging at the same rate with the wifi kill switch off and wifi off (about 2,3Watt) it took way more time to get the battery discharge 4% whit the wifi turned mechanically off. 
With the wifi software off, the phone would drop 4% in about 3 minutes and discharge rate is higher (3Watt) when wifi is turned off with the software!
With the wifi kill switches off, the phone dropped 4% in about 11 minutes while idle... This make a huge difference on battery life!

So this is very odd... at least this confirm that there is an issue with wifi power management on the software side...

bfra2373, 2023-08-12:
well sadly, a day later in the same condition (phone idle and wifi kill switche is off) the phone drop 1%/minutes with only powersupply app running... so maybe this is not wifi related afterall...

maxrdz, 2024-04-20:
Hopefully this comment helps someone in the future, but in my case I ran `top` and filtered by CPU time to see what process was taking up so much battery;

I found out my Bluetooth service constantly keeps pinging / gets pinged by other bt devices forever and consumes all your battery until the service is turned off. Just a note in case this helps someone in the future.