dreemurrs-embedded/Pine64-Arch #573: pinetab2 usb multiplexers don't work.

dozadoesit, 2023-08-02:
- Device: pinetab2
- Kernel Version : 6.4.3-danctnix1-1-pinetab2
- UI: KDE Plasma

### Steps to reproduce
Plug in a USB multiplexer and none of the device on it are recognized

### Expected behavior
usb-c multiplexers should just work like plugging something natively into the pinetab

### Actual behavior
None of the devices plugged in are recognized and eventually the tablet freezes.

Danct12, 2023-08-29:
I don't seem to be able to reproduce this issue, which USB-C hub are you using?

dozadoesit, 2023-08-29:
I am using the HUB 1C3A from ANKER as well as well as this [cable creations](https://www.amazon.com/gp/product/B08FWMWGTD/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1) model. If I plug them into the bottom usb-c port I get nothing at all even the indicator light doesn't light up. if I plug them into the top USB-C port I get indicator lights and the cable creations one emits a concerning audible humming sound. When I plug devices into these hubs they don't power on and they don't register to the OS. Even my yubikeys which are pretty low powered don't work. However if I plug them directly into the USB port without the hub these devices do work. So it seems like ( maybe with just my device ??? ) hubs don't work with the pinetab 2 or maybe just certain hubs don't work.

Danct12, 2023-08-29:
The PinePhone USB hub seems to work for me, I'll test with the PinePhone Pro hub and see if I can reproduce the issue.

Danct12, 2024-07-11:
With 8707e3575c503ce0c20f8bf1eb4f08b01fe73423, the bottom USB-C port should work.

closed 2024-07-24 by Danct12