dreemurrs-embedded/Pine64-Arch #578: VoiceCall UCM profile 'ADC Capture Volume' should be 160

ewtoombs, 2023-08-18:
https://github.com/dreemurrs-embedded/Pine64-Arch/blob/36347aa7735c032f6069156fbec414b800f1e5c8/PKGBUILDS/pine64/alsa-ucm-pinephonepro/VoiceCall.conf#L43

See also https://gitlab.com/pine64-org/pine64-alsa-ucm/-/issues/5

This fixes quiet audio on the other end of a call. 'ADC Mono Capture Volume' should also be 160.

maxrdz, 2024-06-09:
https://github.com/dreemurrs-embedded/Pine64-Arch/blob/36347aa7735c032f6069156fbec414b800f1e5c8/PKGBUILDS/pine64/alsa-ucm-pinephonepro/VoiceCall.conf#L53

This line, which is already in the repo, seems to fix any low audio issues in the internal microphone on the PPP.

For some reason, on my system, `IN1 Boost` was set to `0`, even though my system is up to date and the repo has it set to `8` as seen above.

So, in a nutshell:
```
ADC Capture Volume = 47
ADC Mono Capture Volume = 47
IN1 Boost = 8
```

ewtoombs, 2024-06-10:
This issue recently mysteriously fixed itself on my PPP.
I'm closing it for now.

closed 2024-06-10 by ewtoombs