dreemurrs-embedded/Pine64-Arch #589: PP: Fails to boot with FDE enabled

pentamassiv, 2023-09-18:
**Device info (please complete the following information)**
- Device: PinePhone 
- Kernel Version : 6.4.10-1-danctnix
- UI: Phosh

**Describe the bug**
Ever since I updated on September 15th, the phone does not boot. The following error is displayed:

```
The syntax 'root=/dev/mmcblk2p2' where '/dev/mmcblk2p2' is an encrypted volume is deprecated
Use 'cryptdevice=/dev/mmcblk2p2:root root=/dev/mapper/root' instead
```

I do not get asked for a password to unlock my device and then there are a bunch of more repeating errors

**To Reproduce**
Steps to reproduce the behavior:
1. Update the system
2. Reboot

**Expected behavior**
I should get asked for a password and the phone should boot

**Before submitting a bug report, please be sure that you have done the following:**
- [x] I have checked for updates (pacman -Syu) and installed the updates but the problem persisted.
- [x] I am **not** using Megi's multiboot image, which is unsupported by the developer.

DrRac27, 2023-09-19:
This bug is known since the 2. september in the Matrix room. Some modules needed for osk-sdl are now optional and therefore not in the initramfs anymore. Danct12 said in the matrix room: 
> well crap, that now requires the module to be put in initramfs
> i actually just moved lima and panfrost from built-in to module to save some space
> you can still boot from another media and unlock the drive, mount boot and add "panfrost" to mkinitcpio's modules
> then regen initramfs, that should work

a few days ago I added the module before I updated but I now have the same problem anyways. I have no time to fix that now but maybe it helps.

lily-0xFE5, 2023-09-19:
Making the changes to mkinitcpio.conf and boot.txt from #590 and adding `lima` and `panfrost` to mkinitcpio.conf modules fixed it for me.

Barba732, 2023-10-11:
How to install it with FDE, since the script for it is not working anymore?

realroot2185, 2023-11-13:
I am on Armtix and just adding lima and panfrost to mkinitcpio.conf works (and rebuild it).
The hooks of the PR don't look good, if you use osd-sdl (or unl0kr) you shouldn't use encrypt too.
boot.txt fixes deprecation warning.