dreemurrs-embedded/Pine64-Arch #593: glib2: notification sleep warning

wes1138, 2023-10-15:
**Device info (please complete the following information)**
- Device: PinePhone
- Kernel Version : 6.4.10-1-danctnix
- UI: Phosh

**Describe the bug**
After recent upgrade (ran pacman -Syu on Oct. 14), pinephone appears to receive mystery notifications (notification led blinks, but no notifications show up in the list).  Further, these notifications seem to turn the screen on.  This will happen every few minutes.

**To Reproduce**
Steps to reproduce the behavior:
1. Update system.
2. Let phone rest (press power button, then don't touch anything).
3. In <= 5 minutes, you will get a notification / screen wake up, as described above.

**Expected behavior**
Phone stays asleep in the abscence of real notifications.

**Logfiles and additional information**
Seems like glib2 upgrade is to blame.  The relevant lines in pacman.log:

```
[ALPM] running 'glib-compile-schemas.hook'...
[ALPM-SCRIPTLET] No such key âshow-notification-sleep-warningâ in schema âorg.gnome.settings-daemon.plugins.powerâ as specified in override file â/usr/share/glib-2.0/schemas/11_mobile.gschema.overrideâ; ignoring override for this key.
```

Danct12, 2023-10-15:
Please update your device again. I recently pushed 561b4dd55b9e5bf56502a7d7c30d0ed09849639c which should fix the wake issue.

wes1138, 2023-10-15:
Thank you!  Ran it this morning -- must have just missed your update.

wes1138, 2023-10-15:
Seems resolved.  Thanks so much.

closed 2023-10-15 by wes1138