dreemurrs-embedded/Pine64-Arch #598: p64.arikawa-hi.me SSL UNRECOGNIZED NAME ALERT

Pigpog, 2023-11-18:
- Device: PinePhone Pro
- Kernel Version : 6.5.7-1-danctnix
- UI:  Phosh

**Describe the bug**
When attempting a system update using pacman, it fails to sync danctnix.db due to an SSL error with host p64.arikawa-hi.me

**To Reproduce**
Steps to reproduce the behavior:

`pacman -Sy`

**Expected behavior**
pacman would sync package repository as normal

**Screenshots (pretend it's an image file)**
```
[alarm@pine ~]$ sudo pacman -Syu
[sudo] password for alarm:
:: Synchronizing package databases...
 danctnix.db failed to download
error: failed retrieving file 'danctnix.db' from p64.arikawa-hi.me : OpenSSL/3.1.4: error:0A000458:SSL routines::tlsv1 unrecognized name
error: failed to synchronize all databases (download library error)
```

**Logfiles and additional information**
Immediately before I started getting this error, I had ran pacman -Syu and it sync'd fine. However, it immediately failed with 404 error when trying to download the first package (telegram-desktop)

**Before submitting a bug report, please be sure that you have done the following:**
- [ N/A ] I have checked for updates (pacman -Syu) and installed the updates but the problem persisted.
- [ x ] I am **not** using Megi's multiboot image, which is unsupported by the developer.

Danct12, 2023-11-19:
The old domain name used by the repo was by @diamondburned (thanks for the support) and that is no longer being used.

Open /etc/pacman.conf in vim, nano or any text editor and change `p64.arikawa-hi.me` domain to `archmobile.mirror.danctnix.org`

Alternatively it can be done using sed:
`sudo sed -i 's/p64.arikawa-hi.me/archmobile.mirror.danctnix.org/g' /etc/pacman.conf`

Danct12, 2023-11-19:
I'll close this issue now and pin it, as there's nothing I can do to fix the issue other than making a PSA.

closed 2023-11-19 by Danct12

diamondburned, 2023-11-19:
It just expired today! I went ahead and re-registered the domain; hopefully Namecheap still kept the DNS info. Thanks for making this issue.