dreemurrs-embedded/Pine64-Arch #604: Pinephone will wake up only once

Hunman, 2024-01-03:
**Device info**
- **Device**: PinePhone (regular)
- **Kernel Version**: 6.6.6-1-danctnix (installed at January 1st from the 2023 September 25th image)
- **UI**: SXMO (experienced with Phosh before too)

**Describe the bug**
The pinephone will suspend when left to it's devices. Afterwards pressing the power button will wake it once.

After another period of inactivity, the phone will suspend once again, but this time the power button will not wake it up.

Have to looooong press the power button to shut the device down, then slightly long press to power it on.

**Expected behavior**
The Phone to wake up every time after it goes to sleep :)

**Screenshots**
N/A

**Logfiles and additional information**
The phone is booting from internal eMMC and the keyboard addon is attached.

Journalctl is extracted here:
https://pastabin.displeases.me/bins/9b018533-53fc-4f3e-aa52-e922c907650a

Raw logs if preferred:
After first wake up (journalctl -b):
https://pastabin.displeases.me/files/9b018834-2849-49bc-bab1-1c73b90211a3

After second sleep and reboot (journalctl -b-1):
https://pastabin.displeases.me/files/9b018887-c5db-4759-97c0-78b63b58fb70

Diff:
https://pastabin.displeases.me/files/9b01898d-ff1a-4058-ad9e-aaa5320f308e

**Before submitting a bug report, please be sure that you have done the following:**
- [X] I have checked for updates (pacman -Syu) and installed the updates but the problem persisted.
- [X] I am **not** using Megi's multiboot image, which is unsupported by the developer.

Danct12, 2024-02-11:
I'm unable to reproduce this issue on 6.6.6, as well as 6.6.10 and 6.7.4. But there's a new U-Boot in the testing repo. Can you take a look at it and see if that fixes your issue?

Hunman, 2024-02-11:
This opened a smaller can of worms, as not only the danctnix-testing repo was not present in the pacman config (even as a comment), but the danctnix repo was missing as well.

After adding the danctnix repo, complete system update, testing, adding testing repo and updating uboot (and only uboot), the phone still wakes up only once sadly :(

However, I remembered that a long long while ago, I attempted to install tow-boot (bad idea, because regular pinephone, I know). Could it be that tow-boot has some leftovers somewhere even though internal eMMC has been essentially wiped by `dd` before?

Here is the pacman log of the packages that updated after adding the danctnix repo:
```
[2024-02-11T14:53:27+0100] [PACMAN] Running 'pacman -Syyu'
[2024-02-11T14:53:27+0100] [PACMAN] synchronizing package lists
[2024-02-11T14:53:32+0100] [PACMAN] starting full system upgrade
[2024-02-11T14:54:08+0100] [PACMAN] Running 'pacman -Syi linux-megi'
[2024-02-11T14:54:08+0100] [PACMAN] synchronizing package lists
[2024-02-11T14:55:05+0100] [PACMAN] Running 'pacman -Syyu'
[2024-02-11T14:55:05+0100] [PACMAN] synchronizing package lists
[2024-02-11T14:55:11+0100] [PACMAN] starting full system upgrade
[2024-02-11T14:55:37+0100] [ALPM] transaction started
[2024-02-11T14:55:38+0100] [ALPM] installed conky (1.19.6-1)
[2024-02-11T14:55:38+0100] [ALPM] upgraded conky-wayland (1.18.1-1 -> 1.18.1-99)
[2024-02-11T14:55:38+0100] [ALPM] downgraded libgpiod (2.1-1 -> 1.6.4-1)
[2024-02-11T14:55:38+0100] [ALPM] upgraded eg25-manager (0.4.6-2 -> 0.4.6-4)
[2024-02-11T14:55:38+0100] [ALPM] upgraded mkinitcpio (37.3-1 -> 37.3-1.1)
[2024-02-11T14:55:45+0100] [ALPM] upgraded linux-megi (6.6.6-1 -> 6.6.10-1)
[2024-02-11T14:55:45+0100] [ALPM] upgraded lisgd (0.3.5-1 -> 0.4.0-1)
[2024-02-11T14:55:45+0100] [ALPM] upgraded pnc (0.9.4-3 -> 0.9.4-4)
[2024-02-11T14:55:45+0100] [ALPM] installed brightnessctl (0.5.1-2)
[2024-02-11T14:55:45+0100] [ALPM] upgraded sxmo-utils-sway (1.14.1-2 -> 1.15.2-1)
[2024-02-11T14:55:45+0100] [ALPM] installed libplist (2.3.0-2)
[2024-02-11T14:55:45+0100] [ALPM] installed libusbmuxd (2.0.2-3)
[2024-02-11T14:55:45+0100] [ALPM] installed libimobiledevice (1.3.0-10)
[2024-02-11T14:55:45+0100] [ALPM] installed usbmuxd (1.1.1-3)
[2024-02-11T14:55:45+0100] [ALPM] installed upower (1.90.2-1)
[2024-02-11T14:55:45+0100] [ALPM] warning: directory permissions differ on /usr/share/polki
t-1/rules.d/
filesystem: 755  package: 750
[2024-02-11T14:55:46+0100] [ALPM] upgraded sxmo-utils (1.14.1-2 -> 1.15.2-1)
[2024-02-11T14:55:46+0100] [ALPM-SCRIPTLET] It is recommended you run sxmo_migrate.sh to ch
eck your configuration files and custom hooks against the defaults (it will not make any ch
anges unless explicitly told to)
[2024-02-11T14:55:46+0100] [ALPM] transaction completed
```

Danct12, 2024-02-24:
You can try to dd zero to mmcblk2boot0.

https://github.com/Tow-Boot/Tow-Boot/blob/released/boards/pine64-pinephoneA64/default.nix#L47C28-L47C45

Danct12, 2024-09-28:
Is this still an issue? Please reopen if you can still reproduce the issue!

closed 2024-09-28 by Danct12

Hunman, 2024-09-30:
The issue still persists, but I'm ready to write it off as a hardware issue, as I have some other issues with my device, like the keyboard addon not working and the screen has an insensitive vertical line where touch is not detected

[image]