dreemurrs-embedded/Pine64-Arch #607: Pinephone Pro throwing up errors on sudo pacman -Syu

weevil23, 2024-01-10:
Doing a routing update and it is throwing the following errors:
```:: Starting full system upgrade...
resolving dependencies...
:: There are 2 providers available for dbus-units:
:: Repository core
   1) dbus-broker-units  2) dbus-daemon-units

Enter a number (default=1): 1
looking for conflicting packages...

Packages (24) dbus-1.14.10-2  dbus-broker-35-2  dbus-broker-units-35-2
              eg25-manager-0.4.6-3  enchant-2.6.5-1  ffmpeg-2:6.1.1-1
              firefox-121.0.1-1  folks-0.15.7-1  gnome-control-center-45.2-1
              gnome-shell-1:45.3-1  gperftools-2.15-1  libbsd-0.11.8-1
              libgpiod-2.0.1-1  libgpiod1-1.6.4-1  libluv-1.47.0_0-1
              libplacebo-6.338.1-1  libwbclient-4.19.4-1  pixman-0.43.0-1
              python-pillow-10.2.0-1  smbclient-4.19.4-1  systemd-255.2-2.1
              systemd-libs-255.2-2.1  systemd-sysvcompat-255.2-2.1
              xxhash-0.8.2-1

Total Installed Size:  361.97 MiB
Net Upgrade Size:        2.06 MiB

:: Proceed with installation? [Y/n] y
(24/24) checking keys in keyring                   [######################] 100%
(24/24) checking package integrity                 [######################] 100%
(24/24) loading package files                      [######################] 100%
(24/24) checking for file conflicts                [######################] 100%
error: failed to commit transaction (conflicting files)
/usr/bin/gpiodetect exists in both 'libgpiod1' and 'libgpiod'
/usr/bin/gpioget exists in both 'libgpiod1' and 'libgpiod'
/usr/bin/gpioinfo exists in both 'libgpiod1' and 'libgpiod'
/usr/bin/gpiomon exists in both 'libgpiod1' and 'libgpiod'
/usr/bin/gpioset exists in both 'libgpiod1' and 'libgpiod'
/usr/include/gpiod.h exists in both 'libgpiod1' and 'libgpiod'
/usr/include/gpiod.hpp exists in both 'libgpiod1' and 'libgpiod'
/usr/lib/libgpiod.so exists in both 'libgpiod1' and 'libgpiod'
/usr/lib/libgpiodcxx.so exists in both 'libgpiod1' and 'libgpiod'
/usr/lib/pkgconfig/libgpiod.pc exists in both 'libgpiod1' and 'libgpiod'
/usr/lib/pkgconfig/libgpiodcxx.pc exists in both 'libgpiod1' and 'libgpiod'
/usr/share/man/man1/gpiodetect.1.gz exists in both 'libgpiod1' and 'libgpiod'
/usr/share/man/man1/gpioget.1.gz exists in both 'libgpiod1' and 'libgpiod'
/usr/share/man/man1/gpioinfo.1.gz exists in both 'libgpiod1' and 'libgpiod'
/usr/share/man/man1/gpiomon.1.gz exists in both 'libgpiod1' and 'libgpiod'
/usr/share/man/man1/gpioset.1.gz exists in both 'libgpiod1' and 'libgpiod'
Errors occurred, no packages were upgraded.
[alarm@danctnix ~]$ 
```

closed 2024-01-10 by Danct12

weevil23, 2024-01-10:
```Name            : libgpiod
Version         : 1.6.3-2
Description     : C library and tools for interacting with the linux GPIO
                  character device
Architecture    : aarch64
URL             : https://git.kernel.org/pub/scm/libs/libgpiod/libgpiod.git
Licenses        : LGPL2.1
Groups          : None
Provides        : None
Depends On      : linux-api-headers>=4.8  python
Optional Deps   : None
Required By     : eg25-manager
Optional For    : None
Conflicts With  : libgpiod-git
Replaces        : None
Installed Size  : 356.80 KiB
Packager        : DanctNIX Build System <builder@main-key.danctnix.org>
Build Date      : Fri 31 Dec 2021 10:27:35 UTC
Install Date    : Mon 25 Sep 2023 16:05:37 UTC
Install Reason  : Installed as a dependency for another package
Install Script  : No
Validated By    : Signature
```

weevil23, 2024-01-10:
This is a 1 month old Pinephone Pro I bought new direct from HK when the started taking orders in Dec. It was shipped with Sailfish/Uboot installed as standard. I am running arch from the Micro SD. eg25-manager must have been preinstalled - I note it has not been updated in over a year. 
What is my route forward please?

mikeshenson, 2024-01-10:
Danct12 fixed your issue. Please give him some time to build that package and get it up on the server.

weevil23, 2024-01-10:
I ran sudo pacman -Rdd libgpiod first then pacman -Syi and that has rebuilt and worked OK

commit 5398e3a7 (2024-01-10): pkg: danctnix: eg25-manager: depend on libgpiod<2.0

Close #607