dreemurrs-embedded/Pine64-Arch #609: No voice in calls, unless configuration is manually switched

lajonss, 2024-02-07:
**Device info (please complete the following information)**
- Device: PinePhone, PinePhone Pro
- Kernel Version : 6.6.10-1-danctnix
- UI: Phosh

**Describe the bug**
Once answered, calls have no sound.

**To Reproduce**
Steps to reproduce the behavior:
(assuming open modem firmware)
1. text modem "call me"
2. answer the call
3. there is no sound

**Expected behavior**
Calls should just work™

**Logfiles and additional information**
Can be worked around by manually selecting proper Voice Call profile in sound settings. This probably has something to do with [PulseAudio 17.0](https://www.freedesktop.org/wiki/Software/PulseAudio/Notes/17.0/), as there are some changes to how Alsa UCM mapping is handled. I'd assume that Callaudiod is not compatible with the new PulseAudio.

Another workaround: downgrading pulseaudio, pulseaudio-bluetooth and libpulse to 16.1.7 seems to help.

**Before submitting a bug report, please be sure that you have done the following:**
- [x] I have checked for updates (pacman -Syu) and installed the updates but the problem persisted.
- [x] I am **not** using Megi's multiboot image, which is unsupported by the developer.

Danct12, 2024-02-11:
I have provided libpulse and pulseaudio 16.1-7 in the repo, running `pacman -Syuu` should downgrade it.

maxrdz, 2024-03-14:
where can I find the phone number that the modem firmware responds to?

maxrdz, 2024-03-28:
> I have provided libpulse and pulseaudio 16.1-7 in the repo, running `pacman -Syuu` should downgrade it.

Downgrading just recently broke my PPP, but I did do this downgrade along with the latest few packages I upgraded just right now. I'm still investigating what went off, but on boot 'Netfilter Manager' fails and I have no internet, no modem, and no audio (just a dummy audio channel)

maxrdz, 2024-04-09:
Should this issue be closed?

lajonss, 2024-07-03:
Same problem happens with pipewire-pulse 1:1.2.0-2.

Danct12, 2024-07-04:
This should now be fixed in 30bb6aa1ffa878e52f933cb01c71d8ed44e1a6c4

If this does not fix it, feel free to reopen this issue and provide logs.