dreemurrs-embedded/Pine64-Arch #617: Task switcher doesn't work on Plasma Mobile

Shatur, 2024-03-20:
**Device info (please complete the following information)**
- Device: PinePhone (Pro)
- Kernel Version : 6.7.7-1-danctnix
- UI: Plasma Mobile

**Describe the bug**
Task switching results in black screen. See [this](https://invent.kde.org/teams/plasma-mobile/issues/-/issues/282) issue for details (it even contains a video).

**Workaround**

It happens because Qt6 not compiled with OpenGLES 2 support. Devs consider it as a Kwin and expect it to fallback.

As a temporary workaround we can compile `qt6-base` with `-DQT_FEATURE_opengles2=ON` like [Alpine does](https://gitlab.alpinelinux.org/alpine/aports/-/blob/master/community/qt6-qtbase/APKBUILD#L81).

Diff for the PKGBUILD from ArchLinux ARM:

```diff
diff --git a/PKGBUILD b/PKGBUILD
index 0f5076c..058dc5b 100644
--- a/PKGBUILD
+++ b/PKGBUILD
@@ -114,6 +114,7 @@ build() {
     -DFEATURE_system_sqlite=ON \
     -DFEATURE_system_xcb_xinput=ON \
     -DQT_FEATURE_sql_ibase=OFF \
+    -DQT_FEATURE_opengles2=ON \
     -DCMAKE_MESSAGE_LOG_LEVEL=STATUS
   cmake --build build
 }
```

Also `@DevinLin` (one of the KDE devs) said the following in Matrix:

> it will probably have better performance to have that flag enabled for pine devices so i'd heavily recommend building qtbase with it

**Before submitting a bug report, please be sure that you have done the following:**
- [x] I have checked for updates (pacman -Syu) and installed the updates but the problem persisted.
- [x] I am **not** using Megi's multiboot image, which is unsupported by the developer.

Danct12, 2024-03-24:
qt6-base-es2 is now available in repo.

Shatur, 2024-03-24:
Closed by 0829f7867acc3d082c16d2244030b0f88a91d8df.

closed 2024-03-24 by Shatur