dreemurrs-embedded/Pine64-Arch #618: autorotation is backwards on pinetab 2

dozadoesit, 2024-03-21:
**Device info (please complete the following information)**
- Device: PineTab2
- Kernel Version : $(uname -r)
- UI: Plasma Desktop

**Describe the bug**
with auto-rotation turned on the screen is in portrait when the tablet is in landscape and the screen in is in landscape when the tablet is in portrait.

**To Reproduce**
Steps to reproduce the behavior:
1. update to the latest plasma release with sudo pacman -Syu
2. turn autorotation on when in tablet mode

**Expected behavior**
Screen rotation should match the orientation of the device

nerdcorenet, 2024-04-18:
> **To Reproduce** Steps to reproduce the behavior:
>     1. update to the latest plasma release with sudo pacman -Syu
>     2. turn autorotation on when in tablet mode

I was experiencing the same issue in recently updated KDE Plasma (libplasma-6.0.3-1 and danctnix-plasma-desktop-settings 20241312-1) when the PineTab2 is disconnected from the keyboard.

I disabled the "Only in tablet mode" option in KDE Plasma but left the "Automatic" orientation enabled (i.e. not "Manual"), and since doing so the incorrect orientation is applied both when the PineTab2 is attached to the keyboard and when it is disconnected from the keyboard equally.

The orientation seems to always be 90 degrees clockwise from what is expected. This means I ned to tilt my head to the right to see the screen properly, because the KDE desktop bottom is drawn to the Left and the desktop top is drawn on the Right.

nerdcorenet, 2024-04-18:
I have set my KDE Plasma to Manual orientation for the time being, and this seems to reflect the nature of the problem well...

[image]
[image]

In the above screenshots you can see that selecting a Landscape orientation will cause the preview above to show Portrait, and vice versa.

Hope this helps.

k8ieone, 2024-12-02:
My PineTab just arrived and had this issue with the factory image. I updated the system, but no change. I also tried installing phosh and the behavior is still the same. Is it possible the sensor could be mounted the wrong way? :sweat_smile:

k8ieone, 2024-12-02:
FYI: Stopping `iio-sensor-proxy` enables manual rotation on phosh, I'm using that for now.

Danct12, 2024-12-12:
I was unable to reproduce the issue with KDE Plasma 6.2.4, seems to rotate correctly for now.

@k8ieone can you tell me your kernel version?

k8ieone, 2024-12-12:
I forgot to post an update but at some point it started working normally. I think I was at 6.11.something when it started working.

Danct12, 2024-12-24:
Seems to be fixed. Closing for now.

closed 2024-12-24 by Danct12