dreemurrs-embedded/Pine64-Arch #619: Latest linux-megi kernel (6.7.7) breaks PPP system

maxrdz, 2024-03-28:
**Device info**
- Device: PinePhone (Pro)
- Kernel Version : 6.7.4-2-danctnix
- UI: Phosh

**Describe the bug**
A systemd service along the name of 'Netfilter Manager' fails on boot.
No internet is available and the modem is not even on (GNOME Firmware shows no modem either)
There is also no audio, just a 'Dummy Audio' output.

**To Reproduce**
Pull from the latest packages.

**Expected behavior**
Work :p

**Logfiles and additional information**
Let me know if you need more info, I am still new to mobile linux devices.

**Before submitting a bug report, please be sure that you have done the following:**
- [x] I have checked for updates (pacman -Syu) and installed the updates but the problem persisted.
- [x] I am **not** using Megi's multiboot image, which is unsupported by the developer.

maxrdz, 2024-03-28:
To clarify, I *believe* this package was the issue, as when I downgraded it along with `libcloudproviders`, everything worked again. I also had upgraded pulseaudio from 16 to 17 again (after the recent downgrade) to make sure PA wasnt an issue.

Danct12, 2024-03-28:
What's the output of `uname -r` and `ls /usr/lib/modules` after the upgrade?

maxrdz, 2024-03-28:
My current `uname -r` is
```
6.7.4-2-danctnix
```

Danct12, 2024-03-29:
Yup. /usr/lib/modules should have a folder with the new kernel version.

Danct12, 2024-04-03:
Yes

maxrdz, 2024-04-03:
> Yes. Please provide output of uname -r and that after the upgrade.

Okay, so with just the kernel upgraded this is the output:
```
6.7.4-2-danctnix
```
That's interesting 🤔

Danct12, 2024-04-03:
Maybe your /boot partition is not mounted? I would check /etc/fstab and see if the mount points were there.

maxrdz, 2024-04-03:
```
# Static information about the filesystems.
# See fstab(5) for details.

# <file system> <dir> <type> <options> <dump> <pass>
UUID=665e5b11-dccd-4a9a-a3d1-d8bc97c3d500	/         	ext4      	rw,relatime	0 1
UUID=1265-1F14      	/boot     	vfat      	rw,relatime,fmask=0022,dmask=0022,codepage=437,iocharset=ascii,shortname=mixed,utf8,errors=remount-ro	0 2
```
I am not familiar with the fstab file, but it looks like it's there if I am reading it right

maxrdz, 2024-04-04:
```
[alarm@max-pinephonepro ~]$ mount | grep \/boot
/dev/mmcblk1p1 on /boot type vfat (rw,relatime,fmask=0022,dmask=0022,codepage=437,iocharset=ascii,shortname=mixed,utf8,errors=remount-ro)
```
Looks like it is mounted

maxrdz, 2024-04-09:
> Could you upload the contents of your `/boot` directory in an archive here?

https://filetransfer.io/data-package/0gnweRlh#link

qiu-x, 2024-04-10:
Thanks, I checked the kernel version in the archive and it's `6.7.4-2-danctnix`
Full version (extracted with `gzip -dc Image.gz | strings | grep 'Linux version'`):
```
Linux version 6.7.4-2-danctnix (builduser@alarm) (aarch64-unknown-linux-gnu-gcc (GCC) 12.1.0, GNU ld (GNU Binutils) 2.38) # SMP PREEMPT_DYNAMIC 
```
Please run the following command and post the output here:
```
# pacman -Sy linux-megi
```
Afterwards, try verifying the kernel version that's installed in `/boot`.
Perhaps there's some kind of issue that prevents the new image from being saved.

maxrdz, 2024-04-11:
I did a full system upgrade, but here is the snippet of the output for `linux-megi`:
```
(11/24) Updating linux-megi module dependencies...
(12/24) Updating linux-megi initcpios...
==> Building image from preset: /etc/mkinitcpio.d/linux-megi.preset: 'default'
==> Using configuration file: '/etc/mkinitcpio.conf'
  -> -k 6.7.7-1-danctnix -c /etc/mkinitcpio.conf -g /boot/initramfs-linux.img
==> Starting build: '6.7.7-1-danctnix'
  -> Running build hook: [base]
  -> Running build hook: [udev]
  -> Running build hook: [autodetect]
  -> Running build hook: [modconf]
  -> Running build hook: [kms]
  -> Running build hook: [keyboard]
  -> Running build hook: [keymap]
  -> Running build hook: [consolefont]
==> WARNING: consolefont: no font found in configuration
  -> Running build hook: [block]
  -> Running build hook: [filesystems]
  -> Running build hook: [fsck]
==> Generating module dependencies
==> Creating gzip-compressed initcpio image: '/boot/initramfs-linux.img'
bsdtar: Failed to set default locale
bsdtar: Failed to set default locale
==> Initcpio image generation successful
(13/24) Updating linux initcpios...
==> Building image from preset: /etc/mkinitcpio.d/linux-megi.preset: 'default'
==> Using configuration file: '/etc/mkinitcpio.conf'
  -> -k 6.7.7-1-danctnix -c /etc/mkinitcpio.conf -g /boot/initramfs-linux.img
==> Starting build: '6.7.7-1-danctnix'
  -> Running build hook: [base]
  -> Running build hook: [udev]
  -> Running build hook: [autodetect]
  -> Running build hook: [modconf]
  -> Running build hook: [kms]
  -> Running build hook: [keyboard]
  -> Running build hook: [keymap]
  -> Running build hook: [consolefont]
==> WARNING: consolefont: no font found in configuration
  -> Running build hook: [block]
  -> Running build hook: [filesystems]
  -> Running build hook: [fsck]
==> Generating module dependencies
==> Creating gzip-compressed initcpio image: '/boot/initramfs-linux.img'
bsdtar: Failed to set default locale
bsdtar: Failed to set default locale
==> Initcpio image generation successful
```
Then, restarted the device and ran the following commands locally (no Wi-FI, no modem loaded):
```
$ su
# cd /boot
# gzip -dc Image.gz | strings | grep 'Linux version'
... 6.7.7-1-danctnix (builduser@alarm) ...
```

I did not have the 6.7.7 megi kernel installed beforehand; I daily drive this device so I install and uninstall it every time I troubleshoot :p For this output, it was installed, though.

Same issues stand; No Wi-fi and no modem loaded.

qiu-x, 2024-04-11:
Ok, so now that we have the correct version in `/boot`, please post the output of `dmesg`, `uname -r` and `ls /usr/lib/modules`. Please make sure that while you run these commands, the new kernel version is installed and booted.

maxrdz, 2024-04-11:
OK, ran the following with the linux-megi 6.7.7 kernel installed and booted.

dmesg:
[https://pastebin.com/1Qms4xZR](https://pastebin.com/1Qms4xZR)
Pass: DDGTWu7X91

uname -r:
```
6.7.4-2-danctnix
```

ls /usr/lib/modules:
```
6.7.7-1-danctnix
extramodules-6.7
```

qiu-x, 2024-04-11:
Hmm, very weird. Please post also the output of `lsblk`

maxrdz, 2024-04-11:
> Hmm, very weird. Please post also the output of `lsblk`

```
NAME         MAJ:MIN RM   SIZE RO TYPE MOUNTPOINTS
mmcblk2      179:0    0 115.2G  0 disk 
├─mmcblk2p1  179:1    0   122M  0 part /efi
└─mmcblk2p2  179:2    0 115.1G  0 part /
mmcblk2boot0 179:32   0     4M  1 disk 
mmcblk2boot1 179:64   0     4M  1 disk 
mmcblk1      179:96   0  29.7G  0 disk 
├─mmcblk1p1  179:97   0   122M  0 part /boot
└─mmcblk1p2  179:98   0  29.6G  0 part 
zram0        254:0    0   1.5G  0 disk [SWAP]
```

qiu-x, 2024-04-12:
Hmm, seems like we might have found some thing related to the issue. You're boot partition is mounted from your SD card, but the root partition if from your EMMC. I'm surprised to see an EFI partition, which bootloader are you using?

Also, please try removing your SD card and then updating the kernel again.

maxrdz, 2024-04-12:
> Hmm, seems like we might have found some thing related to the issue. You're boot partition is mounted from your SD card, but the root partition if from your EMMC. I'm surprised to see an EFI [...]

I didn't even notice that haha, that is very odd. I will upgrade now, but here is `lsblk` output with the SD card out (no upgrade yet)

```
NAME         MAJ:MIN RM   SIZE RO TYPE MOUNTPOINTS
mmcblk2      179:0    0 115.2G  0 disk 
├─mmcblk2p1  179:1    0   122M  0 part /boot
└─mmcblk2p2  179:2    0 115.1G  0 part /
mmcblk2boot0 179:32   0     4M  1 disk 
mmcblk2boot1 179:64   0     4M  1 disk 
zram0        254:0    0   1.5G  0 disk [SWAP]
```

maxrdz, 2024-04-12:
It looks like the audio and modem/wifi is working as usual.

uname -r says `6.7.7-1-danctnix`
ls /usr/lib/modules says `6.7.7-1-danctnix  extramodules-6.7`

Looks like the kernel image is loaded correctly this time.
Thank you for your help and your time. I really appreciate it!

maxrdz, 2024-04-12:
So since my issue was specific to my situation, I will close this issue as its not the maintainer's problem, but I am curious why my boot directory was mounted from my SD card. :p

closed 2024-04-12 by maxrdz

maxrdz, 2024-04-12:
> Hmm, seems like we might have found something related to the issue. You're boot partition is mounted from your SD card, but the root partition if from your EMMC. I'm surprised to see an EFI [...]

Oh I just noticed your question; I am using U-Boot (I believe that's what it's called, it's getting late and I'm too lazy to confirm :p)

qiu-x, 2024-04-12:
IIRC, u-boot's first boot option is the sdcard, so that would explain why it's booting it first (ie. tow-boot is booting from emmc first). 
Out of pure curiosity, would you mind posting the output of `blkid` (ran as root) with the sdcard is plugged in (that is if you didn't format it yet). 
Btw, you wouldn't happen to flash the same OS image to the sdcard and emmc?

maxrdz, 2024-04-12:
> IIRC, u-boot's first boot option is the sdcard, so that would explain why it's booting it first

Ah, that would explain that behavior.

And, yes, I do have the same OS image as the EMMC flashed to the SD card. :P I got a bit confused when I started setting up my PPP 2 months ago, haha.

The output of `blkid` as root with the SD card inserted and device rebooted:
```
/dev/mmcblk2p2: UUID="665e5b11-dccd-4a9a-a3d1-d8bc97c3d500" BLOCK_SIZE="4096" TYPE="ext4" PARTUUID="e413d06c-42ae-4c55-8444-b55de9e8a3a3"
/dev/mmcblk2p1: SEC_TYPE="msdos" UUID="1265-1F14" BLOCK_SIZE="512" TYPE="vfat" PARTLABEL="boot" PARTUUID="84211891-d9d1-416c-9b67-d16bdf5daef4"
/dev/mmcblk1p1: SEC_TYPE="msdos" UUID="1265-1F14" BLOCK_SIZE="512" TYPE="vfat" PARTLABEL="boot" PARTUUID="84211891-d9d1-416c-9b67-d16bdf5daef4"
/dev/mmcblk1p2: UUID="665e5b11-dccd-4a9a-a3d1-d8bc97c3d500" BLOCK_SIZE="4096" TYPE="ext4" PARTUUID="7f4c955a-3077-4879-bef4-a66bf9a9aec1"
/dev/zram0: LABEL="zram0" UUID="9e420062-a9ab-4be9-8b4b-a39a66aae14d" TYPE="swap"
```

qiu-x, 2024-04-13:
Yup, `/dev/mmcblk2p2` and `/dev/mmcblk1p2` have both the same UUID, so that's why it booted from sdcard and then used the emmc as root ;)