dreemurrs-embedded/Pine64-Arch #626: GTK 4 rendering problems

lajonss, 2024-04-09:
**Device info (please complete the following information)**
- Device: PinePhone Pro
- Kernel Version : 6.8.0-2-megi-danctnix
- UI: Phosh

**Describe the bug**
Various menus are rendered with graphical glitches in GTK 4 applications.

**To Reproduce**
Steps to reproduce the behavior:
1. Open Gnome Console (kgx)
2. Open the three-dot menu
3. Witness the glitches

**Screenshots**
Screenshot posted by Shatur on matrix:
[image]

**Logfiles and additional information**
This happens both under 100% and 200% scalling.

Context menu of Gnome Files (nautilus) is also affected.

Downgrading gtk4 package to 1:4.12.5-1 resolves the issue.

Possibly related: [Gnome 46 release notes](https://release.gnome.org/46/)
> Rendering improvements include sharper app interfaces, crisper on-screen text, and clearer UIs when using fractional display scales. These rendering improvements are thanks to GTK’s new renderers, and are found in GNOME apps which use the latest GTK version.

**Before submitting a bug report, please be sure that you have done the following:**
- [x] I have checked for updates (pacman -Syu) and installed the updates but the problem persisted.
- [x] I am **not** using Megi's multiboot image, which is unsupported by the developer.

maxrdz, 2024-04-12:
This issue has also been happening on my device.

[image]

Danct12, 2024-04-15:
I can confirm this issue also.

maxrdz, 2024-05-09:
Has there been more information found on why this issue is happening? Is this an issue that the GTK developers need to resolve, or is it on the distribution maintainers?

qiu-x, 2024-05-09:
> Is this an issue that the GTK developers need to resolve, or is it on the distribution maintainers?

It might also be an issue on the compositor's side. I should be easy to verify if GTK apps also have the rendering bug on sxmo or plasma mobile.

maxrdz, 2024-06-10:
Since I see this issue has been hanging for a while, I have reached out on the GNOME Discourse for help on where to report this issue. See https://discourse.gnome.org/t/gtk-4-rendering-problems-on-mobile-devices-with-phosh/21452

maxrdz, 2024-10-21:
This issue seems to have been fixed since libadwaita 1.5 released. Anyone else also not experiencing issues anymore?