dreemurrs-embedded/Pine64-Arch #630: headphone audio being available in the list while no headphones are plugged in with kernel 6.8

grinceur, 2024-04-19:
**Device info**
- Device: PinePhone 
- Kernel Version : 6.8.0-2-megi-danctnix
- UI: Phosh

**Describe the bug**
the headphones output and input are available in the list of audio input and output even so there is none plupgged in with the kernel in the testing repo.

**To Reproduce**
Steps to reproduce the behavior:
1. install the kernel 6.8.0-2 from testing repo
2. don't plug headphones
3. see headphones in the list of audio devices.

**Expected behavior**
Only have the headphones listed if they are plugged in.

**Screenshots**
[image]

**additionnal infos**
config uses pipewire.

**Before submitting a bug report, please be sure that you have done the following:**
- [x] I have checked for updates (pacman -Syu) and installed the updates but the problem persisted.
- [x] I am **not** using Megi's multiboot image, which is unsupported by the developer.

chookity-pokk, 2024-04-22:
To add to this, I have noticed that it defaults to headphones for audio which is also an issue since no headphones are plugged in.

G-P-L, 2024-04-26:
I also experienced the same issue. Not only it defaults to the headphone, but If I change the output manually to "Internal speaker", after a call were placed, it switches back to the headphone output.
With the latest PipeWire and WirePlumber combo, the long standing audio switching issue after call finally disappeared, instead now I had to fight with this ghost headphone output issue.
As a temporary solution I had to revert to kernel 6.7.7, where everything is working fine using PipeWire.

G-P-L, 2024-06-05:
I just upgraded to kernel 6.9.0 and so far the phantom headphone problem seems to be gone.

chookity-pokk, 2024-06-05:
I've noticed the same thing as @G-P-L. We can likely close this issue for now since it's outdated?

grinceur, 2024-06-06:
the issue isn’t there with 6.9 for me too closing.

closed 2024-06-06 by grinceur