dreemurrs-embedded/Pine64-Arch #640: Pinephone Pro: Conflicting files on system upgrade

maxrdz, 2024-05-29:
**Device info (please complete the following information)**
- Device: PinePhone Pro
- Kernel Version : 6.7.7-1-danctnix
- UI: Phosh

**Describe the bug**
Cannot upgrade without encountering conflicting files on upgrade, see below.

**To Reproduce**
Steps to reproduce the behavior:
1. sudo pacman -Syyu :)

**Expected behavior**
Upgrade without conflicting files.

**Logfiles and additional information**
```
(287/287) checking keys in keyring                                             [#############################################] 100%
(287/287) checking package integrity                                           [#############################################] 100%
(287/287) loading package files                                                [#############################################] 100%
(287/287) checking for file conflicts                                          [#############################################] 100%
error: failed to commit transaction (conflicting files)
glibc-locales: /usr/lib/locale/aa_DJ.utf8/LC_ADDRESS exists in filesystem
glibc-locales: /usr/lib/locale/aa_DJ.utf8/LC_COLLATE exists in filesystem
glibc-locales: /usr/lib/locale/aa_DJ.utf8/LC_CTYPE exists in filesystem
glibc-locales: /usr/lib/locale/aa_DJ.utf8/LC_IDENTIFICATION exists in filesystem
glibc-locales: /usr/lib/locale/aa_DJ.utf8/LC_MEASUREMENT exists in filesystem
glibc-locales: /usr/lib/locale/aa_DJ.utf8/LC_MESSAGES/SYS_LC_MESSAGES exists in filesystem
glibc-locales: /usr/lib/locale/aa_DJ.utf8/LC_MONETARY exists in filesystem
glibc-locales: /usr/lib/locale/aa_DJ.utf8/LC_NAME exists in filesystem
glibc-locales: /usr/lib/locale/aa_DJ.utf8/LC_NUMERIC exists in filesystem
glibc-locales: /usr/lib/locale/aa_DJ.utf8/LC_PAPER exists in filesystem
glibc-locales: /usr/lib/locale/aa_DJ.utf8/LC_TELEPHONE exists in filesystem
```

**Before submitting a bug report, please be sure that you have done the following:**
- [x] I have checked for updates (pacman -Syu) and installed the updates but the problem persisted.
- [x] I am **not** using Megi's multiboot image, which is unsupported by the developer.

Danct12, 2024-05-29:
https://fosstodon.org/@danctnix/112499469468472282

In short, you can append `--overwrite=/usr/lib/locale/\*` to the pacman command and it should upgrade like normal:
```console
# pacman -Syu --overwrite=/usr/lib/locale/\*
```

closed 2024-05-29 by Danct12

maxrdz, 2024-05-29:
Ah okay, was not sure if I should do that. Thank you