dreemurrs-embedded/Pine64-Arch #645: PPP/pipewire/phosh: No sound in calls

lajonss, 2024-06-23:
**Device info (please complete the following information)**
- Device: PinePhone Pro
- Kernel Version : 6.9.0-1-megi-danctnix
- UI: Phosh
- Sound setup: pipewire + pipewire-pulse

**Describe the bug**
There is no sound in calls.

**To Reproduce**
Steps to reproduce the behavior: (assuming open firmware)
1. Text modem "call me"
2. Answer the call

**Logfiles and additional information**
dmesg:
```
rt5640 1-001c: set_dmic_clk rate=12288000 sysclk=12288000 pd=1 idx=4
```

**Before submitting a bug report, please be sure that you have done the following:**
- [x] I have checked for updates (pacman -Syu) and installed the updates but the problem persisted.
- [x] I am **not** using Megi's multiboot image, which is unsupported by the developer.

grinceur, 2024-07-03:
I am having this issue too on my og pinephone running phosh, i notice that when getting a phone call the sound configuration doesn’t change automatically like it used to, but by selecting the `make a phone call` option in configuration in the settings panel, sound works normally.

lajonss, 2024-07-03:
> I am having this issue too on my og pinephone running phosh, i notice that when getting a phone call the sound configuration doesn’t change automatically like it used to, but by selecting the `make a phone call` option in configuration in the settings panel, sound works normally. Also before there where only 2 options available for the `configuration` and now there is 3, could this be the cause ? the switching mechanism could be failing due to too many options available ?

That sounds like https://github.com/dreemurrs-embedded/Pine64-Arch/issues/609 - a different issue.

Just to make sure I had modem call me just now, and sound was working fine. I'll try to figure out if the issue went away or if there are some special circumstances for it to manifset.

grinceur, 2024-07-03:
i am using pipewire btw so should be different…

lajonss, 2024-07-03:
> i am using pipewire btw so should be different…

I've just checked - a recent pipewire update must have triggered the same issue as PulseAudio 17, so it's almost the same issue as #609 
You may find this interesting if you'd like to switch to PulseAudio: https://nicholasroth.net/2023/07/09/the-pulseaudio-pipewire-frankeinstein-that-enables-good-enough-sound-and-camera-pinephone-pro/

As for #645 - switching to PulseAudio doesn't resolve it.

grinceur, 2024-07-04:
issue solved for me with last update, many thanks @Danct12 !

Danct12, 2024-07-04:
This should now be fixed in 30bb6aa1ffa878e52f933cb01c71d8ed44e1a6c4

If this does not fix it, feel free to reopen this issue and provide logs.

closed 2024-07-04 by Danct12