dreemurrs-embedded/Pine64-Arch #648: pipewire + phone calls do not work on the pinephone pro.

ewtoombs, 2024-07-17:
**Device info**
- Device: PinePhone (Pro)
- Kernel Version : 6.7.7-1-danctnix
- UI: Phosh

**Describe the bug**
Phone calls start, but are silent both ways.

**To Reproduce**
Steps to reproduce the behavior:
0. vol up + power on
1. connect USB.
2. curl -L 
'https://github.com/dreemurrs-embedded/Pine64-Arch/releases/download/20240326/archlinux-pinephone-pro-phosh-20240326.img.xz' | xzcat | sudo dd of=/dev/<the device node> bs=1M conv=fsync
3. reboot the phone.
4. pacman -Syu
5. reboot the phone.
6. try calling the phone.

**Expected behavior**
I hear call audio.

**Logfiles and additional information**
Switching back to pulseaudio fixes it:
0. pacman -Rdd pipewire-pulse
1. pacman -S pulseaudio-alsa
2. pacman -Rs pipewire-audio pipewire-session-manager
3. Reboot.
4. Call the phone again.

**Before submitting a bug report, please be sure that you have done the following:**
- [ * ] I have checked for updates (pacman -Syu) and installed the updates but the problem persisted.
- [ * ] I am **not** using Megi's multiboot image, which is unsupported by the developer.

lajonss, 2024-07-18:
This issue has been bugging me for some time now, but I was unable to pin the problem down: #645
I'm glad it's not just me. Unfortunately this workaround doesn't seem to fix the problem for me.

Danct12, 2024-07-18:
Try the latest callaudiod from the testing repo (you need to enable it by editing pacman.conf)

ewtoombs, 2024-07-18:
> Try the latest callaudiod from the testing repo (you need to enable it by editing pacman.conf)

I installed callaudiod from [danctnix-testing].
That fixed the first problem, but now I have a different problem.
Say my ppp is phone A and the phone it is phoning is phone B.
Audio gets from B's microphone to A's speaker, though it's a little quiet.
I can't really hear any audio at all coming from A's microphone to B's speaker.
Instead, on B's speaker, I can hear really bad feedback loops.
I think one of the ten thousand audio settings in A's unnecessarily complicated sound card is wrong.

mikeshenson, 2024-09-21:
ewtoombs,
I just did a complete rebuild of my PinePhonePro with Phosh. Here is a link to how I got calls working.
https://forum.pine64.org/showthread.php?tid=19428

Danct12,
Thank you for all the great work you have done and are doing!

ewtoombs, 2024-09-23:
Sounds like you switched to Biktorbg's UCM, @mikeshenson, and it worked.

Except for PinePhonePro.conf, though.
Why didn't you switch it over exactly?

mikeshenson, 2024-09-23:
Simple Answer, I updated the Biktorgj 0.8.0 because I was told it helped a lot with calls.

Long answer:
I have used the PinePhonePro for over a year now. Early on, the Biktorgj 0.7.2 firmware worked really good. I had calls at an OK state. So, in my experience, I have had calls working in different states. Last week, I had to reload the OS on the PinePhonePro. I was having calls issues and I was told the Biktorgj 0.8.0 helped a lot with calls.

Definitions:
"during a call" is any phone call started or received between my PinePhonePro and my wife's Android phone. 
"Freight train" is during a call, the Android phone sounds like the PinePhonePro is on a train, Android phone is extremely loud and full of noise. 
"Echo" is during a call, the Android phone has a clear half second delay an echo. My wife would here herself. 
"DanctNIX VoiceCall.conf" also includes the DanctNIX HiFi.conf, PinePhone.conf
"Biktorgj VoiceCall.conf" also includes the DanctNIX PinePhone.conf and the Biktorgj HiFi.conf, EnableSeq.conf and DisableSeq.conf

In all situations I tested, the PinePhonePro always seamed to be OK. It was always the Android that showed the issues.

Scenario one:
OS: DanctNIX Arch with Phosh (archlinux-pinephone-pro-phosh-20230203.img.xz)
EG25 Firmware: Biktorgj 0.7.2
Audio: Pulseaudio (not pipewire)
DanctNIX VoiceCall.conf
With some slight edits of the VoiceCall.conf the "Freight train" was fixed and there was only a slight whisper of an "echo".

Scenario two:
OS: DanctNIX Arch with Phosh (archlinux-pinephone-pro-phosh-20240326.img.xz)
EG25 Firmware: Biktorgj 0.7.2
Audio: pipewire (not Pulseaudio)
DanctNIX VoiceCall.conf
With some slight edits of the VoiceCall.conf the "Freight train" was fixed and there was a clear and audible "echo".
Biktorgj VoiceCall.conf
With and without edits on  VoiceCall.conf the "Freight train" was an issue.

Scenario three:
OS: DanctNIX Arch with Phosh (archlinux-pinephone-pro-phosh-20240326.img.xz)
EG25 Firmware: Biktorgj 0.8.0
Audio: pipewire (not Pulseaudio)
DanctNIX VoiceCall.conf
With and without edits on VoiceCall.conf the "Freight train" was fixed and there was a clear and audible "echo".
Biktorgj VoiceCall.conf
Setting the PipeWire Mic volume to 0 with alsamixer.
Without edits on VoiceCall.conf the "Freight train" was fixed and the "echo" is fixed. Also, the side volume up and down buttons work during a phone call.

ewtoombs, 2025-06-08:
One year later, and phone calls are still broken on a linux distro designed to work on phones.
I upgraded the EG25 firmware to pinephone_modem_sdk 0.8.0 and I did a pacman -Syyuu.
It's still broken.
This time, when I try calling +1 804 222 1111, I receive no audio until I hit either volume key.
Then, I get about three seconds of audio before it goes mute again.

mikeshenson, 2025-06-09:
> I tried replacing all of ucm with https://github.com/Biktorgj/pine64-alsa-ucm , except for PinePhonePro.conf. No difference.

I am sorry to hear that you are still having issues with the call audio. The call audio on my PinePhonePro still works.

I only have 2 audio outputs. Here is a screen shot of when the phone is idle.
[image]

Here is a screen shot of during a phone call. The phone automatically switches to Earpiece. The volume needs to be low or the other person will hear themselves. 
[image]

ewtoombs, 2025-06-18:
@mikeshenson : which version of the modem firmware are you running?
Did you still have to modify UCM files?