dreemurrs-embedded/Pine64-Arch #649: Initial pacman -Syu fails on glibc-locales.

ewtoombs, 2024-07-18:
**Device info**
- Device: PinePhone (Pro)
- Kernel Version : 6.7.7-1-danctnix
- UI: Phosh

**Describe the bug**
```
error: failed to commit transaction (conflicting files)
glibc-locales: /usr/lib/locale/aa_DJ.utf8/LC_ADDRESS exists in filesystem
glibc-locales: /usr/lib/locale/aa_DJ.utf8/LC_COLLATE exists in filesystem
glibc-locales: /usr/lib/locale/aa_DJ.utf8/LC_CTYPE exists in filesystem
glibc-locales: /usr/lib/locale/aa_DJ.utf8/LC_IDENTIFICATION exists in filesystem
```
This goes on for pages.

**To Reproduce**

1. Insert SD card into PC.
1. `curl -L 
'https://github.com/dreemurrs-embedded/Pine64-Arch/releases/download/20240326/archlinux-pinephone-pro-phosh-20240326.img.xz' | xzcat | sudo dd of=/dev/mmcblkN bs=1M conv=fsync`
1. Power down phone and insert SD card into phone.
1. vol down + power on.
1. Configure the wifi.
1. `pacman -Syu`

**Expected behavior**
The command completes.

**Logfiles and additional information**

```
$ pacman -Qo /usr/lib/locale/
/usr/lib/locale/ is owned by glibc 2.35-5.1
$ pacman -Qo /usr/lib/locale/aa_DJ.utf8/LC_ADDRESS 
error: No package owns /usr/lib/locale/aa_DJ.utf8/LC_ADDRESS
```

The following fixes it:

```
# rm /usr/lib/locale/*
```

**Before submitting a bug report, please be sure that you have done the following:**
- [ * ] I have checked for updates (pacman -Syu) and installed the updates but the problem persisted.
- [ * ] I am **not** using Megi's multiboot image, which is unsupported by the developer.

ewtoombs, 2024-07-18:
I just found this:

```
$ pacman -Ql glibc-locales
glibc-locales /usr/
glibc-locales /usr/lib/
glibc-locales /usr/lib/glibc-locale/
glibc-locales /usr/lib/glibc-locale/aa_DJ.utf8/
glibc-locales /usr/lib/glibc-locale/aa_DJ.utf8/LC_ADDRESS
glibc-locales /usr/lib/glibc-locale/aa_DJ.utf8/LC_COLLATE
glibc-locales /usr/lib/glibc-locale/aa_DJ.utf8/LC_CTYPE
glibc-locales /usr/lib/glibc-locale/aa_DJ.utf8/LC_IDENTIFICATION
...
```

This is before upgrading. There is also this:

```
$ cat /usr/share/libalpm/hooks/60-glibc-locales.hook
[Trigger]
Type = File
Operation = Install
Operation = Upgrade
Target = usr/lib/locale/*

[Action]
Description = Restoring locales
When = PostTransaction
Exec = /bin/bash -c 'cp -a /usr/lib/glibc-locale/*utf8 /usr/lib/locale/'
```

It appears the newest version of glibc-locales handles the files differently. This issue should go away once the new glibc-locales is in the .img .

closed 2024-07-18 by ewtoombs

Danct12, 2024-07-19:
https://github.com/dreemurrs-embedded/Pine64-Arch/discussions/641