dreemurrs-embedded/Pine64-Arch #654: Cannot connect to Wi-Fi network

maxrdz, 2024-08-16:
**Device info (please complete the following information)**
- Device: PinePhone Pro
- Kernel Version : 6.9.7-1-megi-danctnix
- UI: Phosh

**Describe the bug**
Cannot connect to Wi-Fi network.

**To Reproduce**
Unknown, not sure if it was after a system upgrade.

**Logfiles and additional information**
Cannot provide logs since SSH is no available due to the issue.

**Before submitting a bug report, please be sure that you have done the following:**
- [x] I have checked for updates (pacman -Syu) and installed the updates but the problem persisted.
- [x] I am **not** using Megi's multiboot image, which is unsupported by the developer.

maxrdz, 2024-08-16:
Running `doas systemctl status NetworkManager` shows some warns logged by netman.

Two lines seem to address the issue:
```
(wlan0): Activation: (wifi) association took too long
(wlan0): state change: config -> failed (reason 'no-secrets', sys-iface-state: 'managed')

```

Danct12, 2024-08-16:
Add `brcmfmac.feature_disable=0x82000` to the kernel command line and wait for a fix.

maxrdz, 2024-08-17:
> Add `brcmfmac.feature_disable=0x82000` to the kernel command line and wait for a fix.

I tried to do this by editing the GRUB configuration, which should be at `/etc/default/grub`, but no file exists on my system. Which method do you recommend to append this kernel parameter?

maxrdz, 2024-08-17:
Oh I might be clueless. I assume GRUB is not installed on the Pinephone is it :p

maxrdz, 2024-08-17:
> Add `brcmfmac.feature_disable=0x82000` to the kernel command line and wait for a fix.

Tried `doas sysctl brcmfmac.feature_disable=0x82000` to set it at runtime, but it says the file does not exist under `/proc/sys/...`.

Danct12, 2024-08-20:
No, this is a kernel cmdline.

You need to edit boot.txt and run ./mkscr script in /boot.

maxrdz, 2024-08-20:
This workaround fixed my issue, thank you!

Danct12, 2024-08-29:
This workaround is now a part of c2b933bbc4b5c4f1bae4a869c272f6669429fe14

closed 2024-09-21 by maxrdz