dreemurrs-embedded/Pine64-Arch #661: Conflicting files on upgrading `linux-firmware`

maxrdz, 2024-09-27:
**Device info (please complete the following information)**
- Device: PinePhone Pro
- Kernel Version : 6.9.7-1-megi-danctnix
- UI: Barebone / Phosh / Plasma Mobile / Plasma Desktop

**Logfiles and additional information**
error: failed to commit transaction (conflicting files)
linux-firmware: /usr/lib/firmware/rtl_bt/rtl8723cs_xx_config.bin exists in filesystem (owned by rtl8723bt-firmware)
linux-firmware: /usr/lib/firmware/rtl_bt/rtl8723cs_xx_fw.bin exists in filesystem (owned by rtl8723bt-firmware)
Errors occurred, no packages were upgraded.

**Before submitting a bug report, please be sure that you have done the following:**
- [x] I have checked for updates (pacman -Syu) and installed the updates but the problem persisted.
- [x] I am **not** using Megi's multiboot image, which is unsupported by the developer.

Danct12, 2024-09-28:
PinePhone Pro doesn't use RTL8723CS WLAN/BT chip. You can remove it.

closed 2024-09-28 by Danct12

maxrdz, 2024-09-28:
Gotcha. Thank you!

commit f64a9f85 (2024-09-28): pkg: pine64: rtl8723bt-firmware: turn into dummy package

Closes #661