dreemurrs-embedded/Pine64-Arch #662: mmsd-tng, vvmd, purism-chatty: symbol lookup error: undefined symbol: _ZNK4i18n12phonenumbers15PhoneNumberUtil5ParseERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEES9_PNS0_11PhoneNumberE

dekeonus, 2024-09-27:
**Device info**
- Device: PinePhone
- Kernel Version : 6.9.7-1-megi-danctnix
- UI: Phosh

I suspect this is from updating to: libphonenumber-1:8.13.46-1 [Build Date: 2024-09-26T04:12:35 UTC]

**Logfiles and additional information**
`Sep 28 04:36:06 dek-pph-ce mmsdtng[7095]: /usr/bin/mmsdtng: symbol lookup error: /usr/bin/mmsdtng: undefined symbol: _ZNK4i18n12phonenumbers15PhoneNumberUtil5ParseERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEES9_PNS0_11PhoneNumberE
Sep 28 04:36:06 dek-pph-ce vvmd[7096]: /usr/bin/vvmd: symbol lookup error: /usr/bin/vvmd: undefined symbol: _ZNK4i18n12phonenumbers15PhoneNumberUtil5ParseERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEES9_PNS0_11PhoneNumberE
Sep 28 04:36:16 dek-pph-ce mmsdtng[7102]: /usr/bin/mmsdtng: symbol lookup error: /usr/bin/mmsdtng: undefined symbol: _ZNK4i18n12phonenumbers15PhoneNumberUtil5ParseERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEES9_PNS0_11PhoneNumberE
Sep 28 04:36:16 dek-pph-ce vvmd[7103]: /usr/bin/vvmd: symbol lookup error: /usr/bin/vvmd: undefined symbol: _ZNK4i18n12phonenumbers15PhoneNumberUtil5ParseERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEES9_PNS0_11PhoneNumberE
Sep 28 04:36:27 dek-pph-ce mmsdtng[7110]: /usr/bin/mmsdtng: symbol lookup error: /usr/bin/mmsdtng: undefined symbol: _ZNK4i18n12phonenumbers15PhoneNumberUtil5ParseERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEES9_PNS0_11PhoneNumberE
Sep 28 04:36:27 dek-pph-ce vvmd[7111]: /usr/bin/vvmd: symbol lookup error: /usr/bin/vvmd: undefined symbol: _ZNK4i18n12phonenumbers15PhoneNumberUtil5ParseERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEES9_PNS0_11PhoneNumberE
Sep 28 04:36:37 dek-pph-ce mmsdtng[7118]: /usr/bin/mmsdtng: symbol lookup error: /usr/bin/mmsdtng: undefined symbol: _ZNK4i18n12phonenumbers15PhoneNumberUtil5ParseERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEES9_PNS0_11PhoneNumberE
Sep 28 04:36:37 dek-pph-ce vvmd[7119]: /usr/bin/vvmd: symbol lookup error: /usr/bin/vvmd: undefined symbol: _ZNK4i18n12phonenumbers15PhoneNumberUtil5ParseERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEES9_PNS0_11PhoneNumberE
Sep 28 04:36:48 dek-pph-ce mmsdtng[7133]: /usr/bin/mmsdtng: symbol lookup error: /usr/bin/mmsdtng: undefined symbol: _ZNK4i18n12phonenumbers15PhoneNumberUtil5ParseERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEES9_PNS0_11PhoneNumberE
Sep 28 04:36:48 dek-pph-ce vvmd[7134]: /usr/bin/vvmd: symbol lookup error: /usr/bin/vvmd: undefined symbol: _ZNK4i18n12phonenumbers15PhoneNumberUtil5ParseERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEES9_PNS0_11PhoneNumberE
Sep 28 04:36:57 dek-pph-ce sm.puri.Chatty.desktop[7146]: /usr/bin/chatty: symbol lookup error: /usr/bin/chatty: undefined symbol: _ZNK4i18n12phonenumbers15PhoneNumberUtil5ParseERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEES9_PNS0_1
1PhoneNumberE`

sorry-i-am-late, 2024-09-28:
I assume this would be the reason why chatty stopped opening and i had to install via flatpak.

Danct12, 2024-09-28:
Should be fixed now.

closed 2024-09-28 by Danct12

dekeonus, 2024-09-28:
purism-chatty still needs to be updated
```console
$ ps -fe|grep -E '[v]vmd|[m]msdtng'
alarm       3782    3770  0 18:24 ?        00:00:00 /usr/bin/mmsdtng -d
alarm       3783    3770  0 18:24 ?        00:00:00 /usr/bin/vvmd -d
```
the mmsd / vvmd are now running
but:
```log
Sep 28 18:26:09 dek-pph-ce sm.puri.Chatty.desktop[4571]: /usr/bin/chatty: symbol lookup error: /usr/bin/chatty: undefined symbol: _ZNK4i18n12phonenumbers15PhoneNumberUtil5ParseERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEES9_PNS0_11PhoneNumberE
```
purism-chatty is still linked to the old libphonenumber

dekeonus, 2024-09-28:
the PKGBUILD for purism-chatty may need to be updated, the upstream Gnome Chatty depends on libphonenumber:
from src/meson.build line 103
```meson
libphonenumber_dep = cc.find_library('phonenumber', required: true)
```
https://gitlab.gnome.org/World/Chatty/-/blob/main/src/meson.build#L103

Danct12, 2024-09-28:
Sorry for that, this should now be fixed.

dekeonus, 2024-09-28:
> Regarding libphonenumber, it's a dependency of evolution-data-server.

As I mentioned above, it is _(now)_ a direct dependency of chatty:
see `src/chatty-phone-utils.cpp` in the chatty source.

https://gitlab.gnome.org/World/Chatty/-/blob/main/src/chatty-phone-utils.cpp

dekeonus, 2024-09-28:
confirming `purism-chatty` is now working.

maxrdz, 2024-09-29:
It is also working for me now. :+1: