dreemurrs-embedded/Pine64-Arch #666: gnome calls: outbound calls only work if its the fisrt call the app has made since start. gsd media keys: can't get default sink/source

budgie-dev, 2024-10-12:
**Device info (please complete the following information)**
- Device: PinePhone
- Kernel Version : 6.11.0-1-megi-danctnix
- UI: Phosh

**Describe the bug**
I noticed that I have the problem described in #648 so i read the bug report and applied the solution. Now calls are only partially working. Only the first outbound call since pkill gnome-calls && gnome-calls works normally. all of the calls give gsd-media-keys errors

**To Reproduce**
Steps to reproduce the behavior:
1. Have the problem described in #648 
2. Apply the solution to #648 
3. Now only the first outbound call since opening gnome-calls works

**Expected behavior**
I should be able to call people

**Screenshots**
Not applicable

**Logfiles and additional information**
First call:
Oct 12 13:45:02 danctnix ModemManager[3698]: <msg> [modem0/call6] user request to start call
Oct 12 13:45:02 danctnix ModemManager[3698]: <msg> [modem0/call6] call state changed: unknown -> dialing (outgoing-started)
Oct 12 13:45:02 danctnix gsd-media-keys[3935]: Unable to get default sink
Oct 12 13:45:02 danctnix gsd-media-keys[3935]: Unable to get default source
Oct 12 13:45:02 danctnix ModemManager[3698]: <msg> [modem0/call6] call is started
Oct 12 13:45:08 danctnix gnome-contacts[5312]: g_object_ref: assertion 'G_IS_OBJECT (object)' failed
Oct 12 13:45:08 danctnix gnome-contacts[5312]: invalid (NULL) pointer instance
Oct 12 13:45:08 danctnix gnome-contacts[5312]: g_signal_handlers_disconnect_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed
Oct 12 13:45:08 danctnix gnome-contacts[5312]: invalid (NULL) pointer instance
Oct 12 13:45:08 danctnix gnome-contacts[5312]: g_signal_handlers_disconnect_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed
Oct 12 13:45:08 danctnix systemd[3732]: dbus-:1.2-org.gnome.Contacts@5.service: Consumed 1.780s CPU time, 11.6M memory peak.
Oct 12 13:45:08 danctnix ModemManager[3698]: <msg> [modem0/call6] call state changed: dialing -> ringing-out (unknown)
Oct 12 13:45:16 danctnix ModemManager[3698]: <msg> [modem0/call6] call state changed: ringing-out -> active (unknown)
Oct 12 13:45:23 danctnix ModemManager[3698]: <msg> [modem0/call6] call state changed: active -> terminated (unknown)
Oct 12 13:45:23 danctnix gsd-media-keys[3935]: Unable to get default sink
Oct 12 13:45:23 danctnix gsd-media-keys[3935]: Unable to get default source

Second call:
Oct 12 13:50:33 danctnix ModemManager[3698]: <msg> [modem0/call8] user request to start call
Oct 12 13:50:33 danctnix ModemManager[3698]: <msg> [modem0/call8] call state changed: unknown -> dialing (outgoing-started)
Oct 12 13:50:33 danctnix gsd-media-keys[3935]: Unable to get default sink
Oct 12 13:50:33 danctnix gsd-media-keys[3935]: Unable to get default source
Oct 12 13:50:33 danctnix ModemManager[3698]: <msg> [modem0/call8] call is started
Oct 12 13:50:35 danctnix ModemManager[3698]: <msg> [modem0/call8] call state changed: dialing -> terminated (unknown)
Oct 12 13:50:35 danctnix gsd-media-keys[3935]: Unable to get default sink
Oct 12 13:50:35 danctnix gsd-media-keys[3935]: Unable to get default source
Oct 12 13:50:39 danctnix gnome-contacts[5470]: g_object_ref: assertion 'G_IS_OBJECT (object)' failed
Oct 12 13:50:39 danctnix gnome-contacts[5470]: invalid (NULL) pointer instance
Oct 12 13:50:39 danctnix gnome-contacts[5470]: g_signal_handlers_disconnect_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed
Oct 12 13:50:39 danctnix gnome-contacts[5470]: invalid (NULL) pointer instance
Oct 12 13:50:39 danctnix gnome-contacts[5470]: g_signal_handlers_disconnect_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed
Oct 12 13:50:39 danctnix systemd[3732]: dbus-:1.2-org.gnome.Contacts@6.service: Consumed 1.555s CPU time, 11.7M memory peak.

**Before submitting a bug report, please be sure that you have done the following:**
- [*] I have checked for updates (pacman -Syu) and installed the updates but the problem persisted.
- [*] I am **not** using Megi's multiboot image, which is unsupported by the developer.

budgie-dev, 2024-10-12:
fixed it partially by adding a custom phone app launcher in phosh:
kill -f '/usr/bin/sh /home/alarm/scripts/calls.sh'
pkill 'gnome-calls %u'
pkill gnome-calls
gnome-calls %u

It sort of works, but for some reason, if you dont enter the number you want to call in around 10 seconds, the error persists.

mikeshenson, 2024-10-14:
"No outbound calls work on pipewire, switch to pulseaudio"
When I took the lastest image and flashed my PinePhonePro, I needed to install "pipewire-alsa" to get the headset speaker and mic to work.
sudo pacman -S pipewire-alsa