dreemurrs-embedded/Pine64-Arch #670: ttyl login loop with Phosh

sorry-i-am-late, 2024-11-01:
- Device: PinePhone  (3/32gb daily diriver)
- Kernel Version : 6.9.7-1
- UI: Phosh

**Describe the bug**
When attempting to boot it does a ttyl login request loop but loops every few seconds and doesnt allow one to actually login even with a external keyboard.  This behavior happens after getting to: "reached target: graphical interface".

Presumably this is just a bad update But I am unsure what package to downgrade when chrooting.

cmoa30, 2024-11-01:
Same  on PinePhonePro, updated this day, reboot, trying to downgrade some installed without solving issue :
- mesa
- xorg-server-common
- xorg-xwayland
- xdg-desktop-portal-gnome

sorry-i-am-late, 2024-11-01:
> Same on PinePhonePro, 

Good to know, I guess that means I should not update my pinetab2 ATM.

kennylevinsen, 2024-11-03:
https://gitlab.gnome.org/World/Phosh/phosh/-/merge_requests/1566

Danct12, 2024-11-05:
Fixed in https://github.com/dreemurrs-embedded/Pine64-Arch/commit/6503f96bf0f510b6c0e6f35e3982f2c49bcee5eb

closed 2024-11-05 by Danct12

maxrdz, 2024-11-05:
> Fixed in [6503f96](https://github.com/dreemurrs-embedded/Pine64-Arch/commit/6503f96bf0f510b6c0e6f35e3982f2c49bcee5eb)

Thank you!