dreemurrs-embedded/Pine64-Arch #671: firefox 132.0-1 dumps core on pinephone

fedov, 2024-11-03:
**Device info (please complete the following information)**
- Device: PinePhone
- Kernel Version : 6.11.0-1-megi-danctnix
- UI: Phosh

**Describe the bug**
firefox 132.0-1 dumps core. Previous versions work fine

**To Reproduce**
Steps to reproduce the behavior:
start firefox

**Logfiles and additional information**

systemd-coredump[15954]: [🡕] Process 15951 (firefox) of user 1000 dumped core.

                                               Stack trace of thread 15952:
                                               #0  0x0000aaaad2cfd000 n/a (firefox + 0x3d000)
                                               #1  0x0000aaaad2cf2204 n/a (firefox + 0x32204)
                                               #2  0x0000aaaad2cf95a0 n/a (firefox + 0x395a0)
                                               #3  0x0000aaaad2cf9230 n/a (firefox + 0x39230)
                                               #4  0x0000ffffb58b3f54 _IO_fclose (libc.so.6 + 0x73f54)
                                               #5  0x0000aaaad2d44fc4 n/a (firefox + 0x84fc4)
                                               #6  0x0000aaaad2d44e3c n/a (firefox + 0x84e3c)
                                               #7  0x0000ffffb58ca624 n/a (libc.so.6 + 0x8a624)
                                               #8  0x0000ffffb593064c n/a (libc.so.6 + 0xf064c)

                                               Stack trace of thread 15951:
                                               #0  0x0000ffffb58c6b9c n/a (libc.so.6 + 0x86b9c)
                                               #1  0x0000ffffb58cc178 n/a (libc.so.6 + 0x8c178)
                                               #2  0x0000aaaad2d44d68 _ZN7mozilla9TimeStamp20ComputeProcessUptimeEv (firefox + 0x84d68)
                                               #3  0x0000aaaad2d449a4 _ZN7mozilla9TimeStamp15ProcessCreationEv (firefox + 0x849a4)
                                               #4  0x0000aaaad2d0ecd8 _ZN7mozilla12baseprofiler13profiler_initEPv (firefox + 0x4ecd8)
                                               #5  0x0000aaaad2cef29c n/a (firefox + 0x2f29c)
                                               #6  0x0000ffffb5868294 n/a (libc.so.6 + 0x28294)
                                               #7  0x0000ffffb5868378 __libc_start_main (libc.so.6 + 0x28378)
                                               #8  0x0000aaaad2cef070 _start (firefox + 0x2f070)
                                               ELF object binary architecture: AARCH64

**Before submitting a bug report, please be sure that you have done the following:**
- [ x] I have checked for updates (pacman -Syu) and installed the updates but the problem persisted.
- [x ] I am **not** using Megi's multiboot image, which is unsupported by the developer.

dekeonus, 2024-11-13:
bug still present in firefox-132.0.1-1

```shell
$ pacman -Qi firefox
Name            : firefox
Version         : 132.0.1-1
Description     : Fast, Private & Safe Web Browser
Architecture    : aarch64
URL             : https://www.mozilla.org/firefox/
Licenses        : MPL-2.0
Groups          : None
Provides        : None
Depends On      : alsa-lib  at-spi2-core  bash  cairo  dbus  ffmpeg  fontconfig  freetype2  gcc-libs  gdk-pixbuf2
                  glib2  glibc  gtk3  hicolor-icon-theme  libpulse  libx11  libxcb  libxcomposite  libxdamage  libxext
                  libxfixes  libxrandr  libxss  libxt  mime-types  nspr  nss  pango  ttf-font
Optional Deps   : hunspell-en_US: Spell checking, American English
                  libnotify: Notification integration [installed]
                  networkmanager: Location detection via available WiFi networks [installed]
                  speech-dispatcher: Text-to-Speech
                  xdg-desktop-portal: Screensharing with Wayland [installed]
Required By     : mobile-config-firefox
Optional For    : None
Conflicts With  : None
Replaces        : None
Installed Size  : 215.61 MiB
Packager        : Arch Linux ARM Build System <builder+seattle@archlinuxarm.org>
Build Date      : 2024-11-06T22:35:49 AEDT
Install Date    : 2024-11-13T11:20:15 AEDT
Install Reason  : Installed as a dependency for another package
Install Script  : No
Validated By    : Signature
```

```shell
$ coredumpctl dump 8228 --output -
           PID: 8228 (firefox)
           UID: 1000 (alarm)
           GID: 1000 (alarm)
        Signal: 11 (SEGV)
     Timestamp: Wed 2024-11-13 11:21:16 AEDT (3min 20s ago)
  Command Line: /usr/lib/firefox/firefox
    Executable: /usr/lib/firefox/firefox
 Control Group: /user.slice/user-1000.slice/user@1000.service/app.slice/app-gnome-firefox-8228.scope
          Unit: user@1000.service
     User Unit: app-gnome-firefox-8228.scope
         Slice: user-1000.slice
     Owner UID: 1000 (alarm)
       Boot ID: e9d4a680ed124416a3ab2e852029643d
    Machine ID: 86229780b302434388a30bf86b59e0d1
      Hostname: dek-pph-ce
       Storage: /var/lib/systemd/coredump/core.firefox.1000.e9d4a680ed124416a3ab2e852029643d.8228.1731457276000000.zst (
present)
  Size on Disk: 1.4M
       Message: Process 8228 (firefox) of user 1000 dumped core.
                
                Stack trace of thread 8228:
                #0  0x0000aaaab261d000 n/a (firefox + 0x3d000)
                #1  0x0000aaaab261228c n/a (firefox + 0x3228c)
                #2  0x0000aaaab26195a0 n/a (firefox + 0x395a0)
                #3  0x0000aaaab2619230 n/a (firefox + 0x39230)
                #4  0x0000ffffb7d42f60 n/a (libwayland-cursor.so.0 + 0x2f60)
                #5  0x0000ffffb7d427ac n/a (libwayland-cursor.so.0 + 0x27ac)
                #6  0x0000ffffb7d41638 wl_cursor_theme_load (libwayland-cursor.so.0 + 0x1638)
                #7  0x0000ffffb8b4d020 gdk_wayland_display_set_cursor_theme (libgdk-3.so.0 + 0x7d020)
                #8  0x0000ffffb8b4d250 n/a (libgdk-3.so.0 + 0x7d250)
                #9  0x0000ffffb8b4cd1c n/a (libgdk-3.so.0 + 0x7cd1c)
                #10 0x0000ffffb8b0fb58 gdk_display_manager_open_display (libgdk-3.so.0 + 0x3fb58)
                #11 0x0000ffffb29d529c n/a (libxul.so + 0x5fd529c)
                #12 0x0000ffffb29d9524 n/a (libxul.so + 0x5fd9524)
                #13 0x0000ffffb29d9a54 n/a (libxul.so + 0x5fd9a54)
[... 41 lines cut ...]
                #2  0x0000ffffb896a794 n/a (libglib-2.0.so.0 + 0x2a794)
                #3  0x0000ffffb89e4b88 n/a (libglib-2.0.so.0 + 0xa4b88)
                #4  0x0000ffffb89e433c n/a (libglib-2.0.so.0 + 0xa433c)
                #5  0x0000ffffbacca624 n/a (libc.so.6 + 0x8a624)
                #6  0x0000ffffbad3064c n/a (libc.so.6 + 0xf064c)
                
                Stack trace of thread 8238:
                #0  0x0000ffffbad2629c ppoll (libc.so.6 + 0xe629c)
                #1  0x0000ffffb89ac2a0 n/a (libglib-2.0.so.0 + 0x6c2a0)
                #2  0x0000ffffb89acc44 g_main_context_iteration (libglib-2.0.so.0 + 0x6cc44)
                #3  0x0000ffffac2ccccc n/a (libdconfsettings.so + 0xcccc)
                #4  0x0000ffffb89e433c n/a (libglib-2.0.so.0 + 0xa433c)
                #5  0x0000ffffbacca624 n/a (libc.so.6 + 0x8a624)
                #6  0x0000ffffbad3064c n/a (libc.so.6 + 0xf064c)
                
                Stack trace of thread 8235:
                #0  0x0000ffffbad2629c ppoll (libc.so.6 + 0xe629c)
                #1  0x0000ffffb89ac2a0 n/a (libglib-2.0.so.0 + 0x6c2a0)
                #2  0x0000ffffb89acc44 g_main_context_iteration (libglib-2.0.so.0 + 0x6cc44)
                #3  0x0000ffffb89acccc n/a (libglib-2.0.so.0 + 0x6cccc)
                #4  0x0000ffffb89e433c n/a (libglib-2.0.so.0 + 0xa433c)
                #5  0x0000ffffbacca624 n/a (libc.so.6 + 0x8a624)
                #6  0x0000ffffbad3064c n/a (libc.so.6 + 0xf064c)
                ELF object binary architecture: AARCH64
```

fedov, 2024-11-28:
> Issue seems fixed in firefox 133.0-1

same here

closed 2025-01-23 by Danct12