dreemurrs-embedded/Pine64-Arch #672: libtiff-4.7.0-1 breaks megapixels on pinephone

fedov, 2024-11-03:
**Device info (please complete the following information)**
- Device: PinePhone
- Kernel Version : 6.11.0-1-megi-danctnix
- UI: Phosh

**Describe the bug**
Megapixels starts up fine. Taking a picture results in core dump
Works up to libtiff-4.6.0-6

**To Reproduce**
Steps to reproduce the behavior:
1. Take a picture

**Logfiles and additional information**
 Process 22277 (megapixels) of user 1000 dumped core.

                Stack trace of thread 22287:
                #0  0x0000ffff8ee4fdd4 memcpy (libc.so.6 + 0x9fdd4)
                #1  0x0000ffff8f2ac398 n/a (libtiff.so.6 + 0xc398)
                #2  0x0000ffff8f2ae180 TIFFVSetField (libtiff.so.6 + 0xe180)
                #3  0x0000ffff8f2ae294 TIFFSetField (libtiff.so.6 + 0xe294)
                #4  0x0000aaaae0224774 n/a (megapixels + 0x14774)
                #5  0x0000aaaae0225188 n/a (megapixels + 0x15188)
                #6  0x0000aaaae0223608 n/a (megapixels + 0x13608)
                #7  0x0000ffff8f6d986c n/a (libglib-2.0.so.0 + 0x6986c)
                #8  0x0000ffff8f6dc238 n/a (libglib-2.0.so.0 + 0x6c238)
                #9  0x0000ffff8f6dd000 g_main_loop_run (libglib-2.0.so.0 + 0x6d000)
                #10 0x0000aaaae022362c n/a (megapixels + 0x1362c)
                #11 0x0000ffff8ee3a624 n/a (libc.so.6 + 0x8a624)
                #12 0x0000ffff8eea064c n/a (libc.so.6 + 0xf064c)

                Stack trace of thread 22280:
                #0  0x0000ffff8ee9e2a4 syscall (libc.so.6 + 0xee2a4)
                #1  0x0000ffff8f71396c g_cond_wait (libglib-2.0.so.0 + 0xa396c)
                #2  0x0000ffff8f69a818 n/a (libglib-2.0.so.0 + 0x2a818)
                #3  0x0000ffff8f7146dc n/a (libglib-2.0.so.0 + 0xa46dc)
                #4  0x0000ffff8f71433c n/a (libglib-2.0.so.0 + 0xa433c)
                #5  0x0000ffff8ee3a624 n/a (libc.so.6 + 0x8a624)
                #6  0x0000ffff8eea064c n/a (libc.so.6 + 0xf064c)

                Stack trace of thread 22288:
                #0  0x0000ffff90412c7c zbar_scan_y (libzbar.so.0 + 0x12c7c)
                #1  0x0000ffff9041145c n/a (libzbar.so.0 + 0x1145c)
                #2  0x0000ffff90412290 zbar_scan_image (libzbar.so.0 + 0x12290)
                #3  0x0000aaaae0225efc n/a (megapixels + 0x15efc)
                #4  0x0000aaaae0223608 n/a (megapixels + 0x13608)
                #5  0x0000ffff8f6d986c n/a (libglib-2.0.so.0 + 0x6986c)
                #6  0x0000ffff8f6dc238 n/a (libglib-2.0.so.0 + 0x6c238)
                #7  0x0000ffff8f6dd000 g_main_loop_run (libglib-2.0.so.0 + 0x6d000)
                #8  0x0000aaaae022362c n/a (megapixels + 0x1362c)
                #9  0x0000ffff8ee3a624 n/a (libc.so.6 + 0x8a624)
                #10 0x0000ffff8eea064c n/a (libc.so.6 + 0xf064c)
                Stack trace of thread 22282:
                #0  0x0000ffff8ee9629c ppoll (libc.so.6 + 0xe629c)
                #1  0x0000ffff8f6dc2a0 n/a (libglib-2.0.so.0 + 0x6c2a0)
                #2  0x0000ffff8f6dd000 g_main_loop_run (libglib-2.0.so.0 + 0x6d000)
                #3  0x0000ffff8f572610 n/a (libgio-2.0.so.0 + 0x172610)
                #4  0x0000ffff8f71433c n/a (libglib-2.0.so.0 + 0xa433c)
                #5  0x0000ffff8ee3a624 n/a (libc.so.6 + 0x8a624)
                #6  0x0000ffff8eea064c n/a (libc.so.6 + 0xf064c)

                Stack trace of thread 22281:
                #0  0x0000ffff8ee9629c ppoll (libc.so.6 + 0xe629c)
                #1  0x0000ffff8f6dc2a0 n/a (libglib-2.0.so.0 + 0x6c2a0)
                #2  0x0000ffff8f6dcc44 g_main_context_iteration (libglib-2.0.so.0 + 0x6cc44)
                #3  0x0000ffff8f6dcccc n/a (libglib-2.0.so.0 + 0x6cccc)
                #4  0x0000ffff8f71433c n/a (libglib-2.0.so.0 + 0xa433c)
                #5  0x0000ffff8ee3a624 n/a (libc.so.6 + 0x8a624)
                #6  0x0000ffff8eea064c n/a (libc.so.6 + 0xf064c)

                Stack trace of thread 22283:
                #0  0x0000ffff8ee9e2a4 syscall (libc.so.6 + 0xee2a4)
                #1  0x0000ffff8f71396c g_cond_wait (libglib-2.0.so.0 + 0xa396c)
                #2  0x0000ffff8f69a818 n/a (libglib-2.0.so.0 + 0x2a818)
                #3  0x0000ffff8f69ae30 g_async_queue_pop (libglib-2.0.so.0 + 0x2ae30)
                #4  0x0000ffff8e68d508 n/a (libpangoft2-1.0.so.0 + 0xd508)
                #5  0x0000ffff8f71433c n/a (libglib-2.0.so.0 + 0xa433c)
                #6  0x0000ffff8ee3a624 n/a (libc.so.6 + 0x8a624)
                #7  0x0000ffff8eea064c n/a (libc.so.6 + 0xf064c)

                Stack trace of thread 22289:
                #0  0x0000ffff8ee9629c ppoll (libc.so.6 + 0xe629c)
                #1  0x0000ffff8f6dc2a0 n/a (libglib-2.0.so.0 + 0x6c2a0)
                #2  0x0000ffff8f6dd000 g_main_loop_run (libglib-2.0.so.0 + 0x6d000)
                #3  0x0000aaaae022362c n/a (megapixels + 0x1362c)
                #4  0x0000ffff8ee3a624 n/a (libc.so.6 + 0x8a624)
                #5  0x0000ffff8eea064c n/a (libc.so.6 + 0xf064c)

                Stack trace of thread 22284:
                #0  0x0000ffff8ee9629c ppoll (libc.so.6 + 0xe629c)
                #1  0x0000ffff8f6dc2a0 n/a (libglib-2.0.so.0 + 0x6c2a0)
                #2  0x0000ffff8f6dcc44 g_main_context_iteration (libglib-2.0.so.0 + 0x6cc44)
                #3  0x0000ffff8aeccccc n/a (libdconfsettings.so + 0xcccc)
                #4  0x0000ffff8f71433c n/a (libglib-2.0.so.0 + 0xa433c)
                #5  0x0000ffff8ee3a624 n/a (libc.so.6 + 0x8a624)
                #6  0x0000ffff8eea064c n/a (libc.so.6 + 0xf064c)
                Stack trace of thread 22302:
                #0  0x0000ffff8ee36b9c n/a (libc.so.6 + 0x86b9c)
                #1  0x0000ffff8ee39828 pthread_cond_wait (libc.so.6 + 0x89828)
                #2  0x0000ffff693c18d4 n/a (libgallium-24.2.5-arch1.1.so + 0x1c18d4)
                #3  0x0000ffff6939ff0c n/a (libgallium-24.2.5-arch1.1.so + 0x19ff0c)
                #4  0x0000ffff693c1764 n/a (libgallium-24.2.5-arch1.1.so + 0x1c1764)
                #5  0x0000ffff8ee3a624 n/a (libc.so.6 + 0x8a624)
                #6  0x0000ffff8eea064c n/a (libc.so.6 + 0xf064c)

                Stack trace of thread 22277:
                #0  0x0000ffff8ee9629c ppoll (libc.so.6 + 0xe629c)
                #1  0x0000ffff8f6dc2a0 n/a (libglib-2.0.so.0 + 0x6c2a0)
                #2  0x0000ffff8f6dcc44 g_main_context_iteration (libglib-2.0.so.0 + 0x6cc44)
                #3  0x0000ffff8f52120c g_application_run (libgio-2.0.so.0 + 0x12120c)
                #4  0x0000aaaae02174c0 n/a (megapixels + 0x74c0)
                #5  0x0000ffff8edd8294 n/a (libc.so.6 + 0x28294)
                #6  0x0000ffff8edd8378 __libc_start_main (libc.so.6 + 0x28378)
                #7  0x0000aaaae0217530 n/a (megapixels + 0x7530)

                Stack trace of thread 22348:
                #0  0x0000ffff8ee9e2a4 syscall (libc.so.6 + 0xee2a4)
                #1  0x0000ffff8f713fb8 g_cond_wait_until (libglib-2.0.so.0 + 0xa3fb8)
                #2  0x0000ffff8f69a794 n/a (libglib-2.0.so.0 + 0x2a794)
                #3  0x0000ffff8f69b028 g_async_queue_timeout_pop (libglib-2.0.so.0 + 0x2b028)
                #4  0x0000ffff8f714a78 n/a (libglib-2.0.so.0 + 0xa4a78)
                #5  0x0000ffff8f71433c n/a (libglib-2.0.so.0 + 0xa433c)
                #6  0x0000ffff8ee3a624 n/a (libc.so.6 + 0x8a624)
                #7  0x0000ffff8eea064c n/a (libc.so.6 + 0xf064c)
                ELF object binary architecture: AARCH64

**Before submitting a bug report, please be sure that you have done the following:**
- [ x] I have checked for updates (pacman -Syu) and installed the updates but the problem persisted.
- [ x] I am **not** using Megi's multiboot image, which is unsupported by the developer.

G-P-L, 2025-01-12:
Megapixels 1.8.3 released which is solving this issue. Please, update the package in the repository.  
https://gitlab.com/megapixels-org/Megapixels/-/tags/1.8.3

Danct12, 2025-01-14:
Fixed in commit 1976507ff46e2ee061a52d68660131faccf51ce7

closed 2025-01-14 by Danct12