dreemurrs-embedded/Pine64-Arch #673: Networking totally gone

sorry-i-am-late, 2024-11-04:
- Device: PinePhone 
- Kernel Version : 6.11.1
- UI: Barebone / Phosh 

Networking is totally gone as if the hardware doesn't exist but when I boot from an sd card it works fine.  NetworkManger and ModemManager are running in the background but dont seem to be doing anything. 

This happened after I updated via chroot due to the seatd issue. It behaves like it did when the wifi chip fried on my v1.2 mainboard except this time it is also the modem and networking is fine when booting from a not up to date sd card.

sorry-i-am-late, 2024-11-04:
Marking as closed because reinstalling 6.11.1 not via chroot seems to have fixed this issue but hopefully if anyone else encounters this issue they wont have to mess around for as long.

closed 2024-11-04 by sorry-i-am-late

Danct12, 2024-11-05:
You need to mount the /boot partition.