dreemurrs-embedded/Pine64-Arch #676: Unable to build packages using pkgbootstrap.sh on Pinephone

mevisc0, 2024-11-12:
**Device info**
- Device: PinePhone
- Kernel Version : 6.9.7-1-megi-danctnix
- UI: Plasma Mobile

**Describe the bug**
As advised by the contributing guidelines, I tried to build a package in Arch Linux Arm (aarch64) on my Pinephone after making changes to a PKGBUILD.
`./pkgbootstrap.sh build --arch aarch64 --repo danctnix PKGBUILDS/plasma/plasma-dialer/`
The package did not build and I got the following error message:
`/usr/bin/makechrootpkg: illegal option -- x`

**To Reproduce**
Steps to reproduce the behavior:
1. Clone this repo onto the Pinephone
2. cd into Pine64-Arch
3. Install devtools
4. Run `./pkgbootstrap.sh build --arch aarch64 --repo danctnix PKGBUILDS/plasma/plasma-dialer/`
5. See error and the makechrootpkg usage guide

**Expected behavior**
The script runs without errors from makechrootpkg.

**Additional information**
makechrootpkg is included in the devtools package. Arch Linux Arm has a devtools package and a devtools-alarm package that is older. Running `makechrootpkg --help` on the Pinephone shows that the -x option does not exist for either devtools or devtools-alarm in Arch Linux Arm. The option does exist in the the version of makechrootpkg that is included in the devtools package available in Arch Linux on x86_64 architecture, so the error is not present on x86_64.

Removing the -x option from the call to makechrootpkg on line 69 of lib/mkpkg.bash resolves the error on aarch64. Perhaps the option could be conditionally added depending on the architecture that the script is being run on?

Danct12, 2024-11-12:
I haven't got around to write documentation for pkgbootstrap, but you need to create a chroot using `./pmbootstrap chroot create` (see `-h`).

As for the devtools package, yes, Arch Linux ARM used an old version which doesn't have that option, but devtools is an `any` architecture package so you can just pull it from Arch Linux and install it.

mevisc0, 2025-01-21:
Ah, thank you for clarifying! I just tried again with that in mind, but ran into different problems. I would appreciate it if you could point me in the right direction. This time, my goal was to update plasma-mobile and plasma-nano from 6.2.4 to 6.2.5.

Here is what I did:
1. Remove the old Arch Linux Arm version of devtools:
`$ sudo pacman -Rsn devtools`

2. Download and install the Arch Linux version of devtools:
I found the link to download devtools on the Arch Linux package page [here](https://archlinux.org/packages/extra/any/devtools/). 
`$ curl -O -J -L https://archlinux.org/packages/extra/any/devtools/download/`
I found instructions on how to install a package from a .zst file [here](https://wiki.archlinux.org/title/Offline_installation#Install_from_file).
`$ sudo pacman -U download`

3. Update the PKGBUILD version, checksum, and .SRCINFO for plasma-mobile:
`$ cd ./Pine64-Arch/PKGBUILDS/plasma/plasma-mobile/`
`$ sed -i 's/pkgver=6\.2\.4$/pkgver=6\.2\.5/' PKGBUILD`
Note: updpkgsums downloads the .tar.xz and .tar.xz.sig files for the package from KDE's servers. I just delete them once the checksums have been updated in PKGBUILD.
`$ updpkgsums PKGBUILD`
I learned about .SRCINFO files and how to update them [here](https://wiki.archlinux.org/title/.SRCINFO#Generation).
`$ makepkg --printsrcinfo > .SRCINFO`

4. Repeat step 3 but for the ./Pine64-Arch/PKGBUILDS/plasma/plasma-mobile/ directory.
5. Create the chroot:
`$ cd ./Pine64-Arch`
`$ ./pkgbootstrap.sh chroot create --arch aarch64 --repo danctnix PKGBUILDS/plasma/plasma-mobile/`
6. Build the package with the updated version:
`$ cd ./Pine64-Arch`
`$ ./pkgbootstrap.sh build --arch aarch64 --repo danctnix PKGBUILDS/plasma/plasma-mobile/`

The command above fails. This is what I am left with:
```
$ ./pkgbootstrap.sh build --arch aarch64 --repo danctnix PKGBUILDS/plasma/plasma-mobile/
[sudo] password for alarm: 
==> Synchronizing chroot copy [/var/lib/danctnix-chroots/danctnix-aarch64/root] -> [alarm]...done
==> ERROR: chroot '/var/lib/danctnix-chroots/danctnix-aarch64/alarm' is not at version v5. Please rebuild.
==> Making package: plasma-mobile 6.2.5-1 (Tue Jan 21 14:27:04 2025)
==> Retrieving sources...
  -> Downloading plasma-mobile-6.2.5.tar.xz...
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
100 3690k  100 3690k    0     0  1617k      0  0:00:02  0:00:02 --:--:-- 2798k
  -> Downloading plasma-mobile-6.2.5.tar.xz.sig...
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   833  100   833    0     0   2221      0 --:--:-- --:--:-- --:--:--  2227
==> Validating source files with sha512sums...
    plasma-mobile-6.2.5.tar.xz ... Passed
    plasma-mobile-6.2.5.tar.xz.sig ... Skipped
==> Verifying source file signatures with gpg...
    plasma-mobile-6.2.5.tar.xz ... FAILED (unknown public key D7574483BB57B18D)
==> ERROR: One or more PGP signatures could not be verified!
==> ERROR: Could not download sources.
```

I am stumped by these errors because I just built the chroot and updated the checksums, so I am not sure how to rebuild the chroot or why there are PGP signature issues. Perhaps there is some flaw in my process, or maybe there are more steps needed to update and build these packages? Thank you for any wisdom you can impart.

Danct12, 2025-01-23:
To obtain the public key, run `gpg --recv-keys D7574483BB57B18D`