dreemurrs-embedded/Pine64-Arch #680: PineTab2 phosh UI rendering broken

k8ieone, 2024-12-02:
## Device info (please complete the following information)
- Device: PineTab2
- Kernel Version : `6.11.4-danctnix1-1-pinetab2`
- UI: Phosh (0.42.0-2)

## Description
GTK4 applications don't render properly. Text and icons are missing. Some icons corrupted.

## Reproduction steps
Steps to reproduce the behavior:
1. Remove everything Plasma-related
2. Install `phosh` and `gnome-control-center`
3. Enable `phosh.service`
4. Try opening settings

## 
The UI should render properly.

## Screenshots

### GNOME Settings
[image]

### Powersupply (works, GTK3)
[image]

### Nautilus - Files
[image]

### Epiphany - GNOME Web
[image]

## Logfiles and additional information
I've also noticed if I let the screen turn off, it'll only come up blank - the phosh lock screen won't come up. I can still connect to the device using SSH. This doesn't always happen.

GTK3 applications seem to work fine - see the second screenshot.

- [x] I have checked for updates (pacman -Syu) and installed the updates but the problem persisted.
- [x] I am **not** using Megi's multiboot image, which is unsupported by the developer.

Danct12, 2024-12-06:
Possibly related to #626.

Try changing `GSK_RENDERER` variable in `/etc/profile.d/gsk-renderer-gl.sh` from `ngl` to `gl`.

closed 2024-12-11 by Danct12

k8ieone, 2024-12-12:
I updated to Phosh 0.43 yesterday and it seems like the lock screen issue seems to be gone. I can lock and unlock as I please :yum:

commit 0b2526dc (2024-12-11): pkg: pine64: device-pine64-pinetab2: switch to gl renderer for gtk

Closes #680