dreemurrs-embedded/Pine64-Arch #681: `hdmi-audio-codec` errors on boot on PPP

maxrdz, 2024-12-03:
**Device info**
- Device: PinePhone Pro
- Kernel Version : 6.11.5-1-megi-danctnix
- UI: Phosh

**Describe the bug**
Sorry, I am not well educated on Linux mobile, but all I have seen that lead me to file this issue is the following:

- Audio is not working properly (+/- volume buttons do not respond, audio outputs sometimes disappear, or default to "Dummy Output", and bluetooth to my car for audio doesnt work either)
- Sometimes the device entirely crashes into a black screen
- Output on boot shows some errors for 'hdmi-audio-codec'.

The following are some errors I found using `doas journalctl --boot=-1 -r`:
```
Dec 03 03:11:02 max-pinephonepro kernel: hdmi-audio-codec hdmi-audio-codec.0.auto: ASoC: error at snd_soc_dai_digital_mute on i2s-hifi: -19
Dec 03 03:11:02 max-pinephonepro kernel: hdmi-audio-codec hdmi-audio-codec.0.auto: ASoC: error at snd_soc_dai_digital_mute on i2s-hifi: -19
Dec 03 03:11:02 max-pinephonepro kernel:  ff8a0000.i2s-i2s-hifi: ASoC: error at __soc_pcm_hw_params on ff8a0000.i2s-i2s-hifi: -19
Dec 03 03:11:02 max-pinephonepro kernel: hdmi-audio-codec hdmi-audio-codec.0.auto: ASoC: error at snd_soc_dai_digital_mute on i2s-hifi: -19
Dec 03 03:11:02 max-pinephonepro kernel: hdmi-audio-codec hdmi-audio-codec.0.auto: ASoC: error at snd_soc_dai_hw_params on i2s-hifi: -19
Dec 03 03:11:02 max-pinephonepro kernel: hdmi-audio-codec hdmi-audio-codec.0.auto: ASoC: error at snd_soc_dai_digital_mute on i2s-hifi: -19
Dec 03 03:11:02 max-pinephonepro kernel:  ff8a0000.i2s-i2s-hifi: ASoC: error at __soc_pcm_hw_params on ff8a0000.i2s-i2s-hifi: -19
Dec 03 03:11:01 max-pinephonepro kernel: hdmi-audio-codec hdmi-audio-codec.0.auto: ASoC: error at snd_soc_dai_digital_mute on i2s-hifi: -19
Dec 03 03:11:01 max-pinephonepro kernel: hdmi-audio-codec hdmi-audio-codec.0.auto: ASoC: error at snd_soc_dai_hw_params on i2s-hifi: -19
Dec 03 03:11:01 max-pinephonepro kernel: hdmi-audio-codec hdmi-audio-codec.0.auto: ASoC: error at snd_soc_dai_digital_mute on i2s-hifi: -19
Dec 03 03:11:01 max-pinephonepro kernel:  ff8a0000.i2s-i2s-hifi: ASoC: error at __soc_pcm_hw_params on ff8a0000.i2s-i2s-hifi: -19
Dec 03 03:11:01 max-pinephonepro kernel: hdmi-audio-codec hdmi-audio-codec.0.auto: ASoC: error at snd_soc_dai_digital_mute on i2s-hifi: -19
Dec 03 03:11:01 max-pinephonepro kernel: hdmi-audio-codec hdmi-audio-codec.0.auto: ASoC: error at snd_soc_dai_hw_params on i2s-hifi: -19
Dec 03 03:11:01 max-pinephonepro kernel: hdmi-audio-codec hdmi-audio-codec.0.auto: ASoC: error at snd_soc_dai_digital_mute on i2s-hifi: -19
Dec 03 03:11:01 max-pinephonepro kernel:  ff8a0000.i2s-i2s-hifi: ASoC: error at __soc_pcm_hw_params on ff8a0000.i2s-i2s-hifi: -19
Dec 03 03:11:01 max-pinephonepro kernel: hdmi-audio-codec hdmi-audio-codec.0.auto: ASoC: error at snd_soc_dai_digital_mute on i2s-hifi: -19
Dec 03 03:11:01 max-pinephonepro kernel: hdmi-audio-codec hdmi-audio-codec.0.auto: ASoC: error at snd_soc_dai_hw_params on i2s-hifi: -19
Dec 03 03:11:01 max-pinephonepro kernel: hdmi-audio-codec hdmi-audio-codec.0.auto: ASoC: error at snd_soc_dai_digital_mute on i2s-hifi: -19
Dec 03 03:11:01 max-pinephonepro kernel:  ff8a0000.i2s-i2s-hifi: ASoC: error at __soc_pcm_hw_params on ff8a0000.i2s-i2s-hifi: -19
Dec 03 03:11:01 max-pinephonepro kernel: hdmi-audio-codec hdmi-audio-codec.0.auto: ASoC: error at snd_soc_dai_digital_mute on i2s-hifi: -19
Dec 03 03:11:01 max-pinephonepro kernel: hdmi-audio-codec hdmi-audio-codec.0.auto: ASoC: error at snd_soc_dai_hw_params on i2s-hifi: -19
```

**To Reproduce**
I am not sure how to reproduce this issue.

**Before submitting a bug report, please be sure that you have done the following:**
- [x] I have checked for updates (pacman -Syu) and installed the updates but the problem persisted.
- [x] I am **not** using Megi's multiboot image, which is unsupported by the developer.

pbvzt3, 2024-12-06:
I can confirm this on a fresh install of the most recent available plasma mobile image (archlinux-pinephone-pro-plasma-20230925.img.xz), updated to current Syu. (Btw, that image is old enough that updating requires selective dependency checking. Will succeed with pacman -Syu --assume-installed plasma-workspace --assume-installed plasma-wayland-session and a subsequent run of pacman -Syu). Was verified with both rk2aw and towboot on spi. Not a hardware issue as old manjaro install on emmc works fine.

System will boot, but gui interface is extremely laggy, to the point that entering pin to log in requires waiting for system to catch up between keypresses, and soon becomes unresponsive or crashes. Terminal access through ssh is workable. Following log in journalctl -f results in a fast, continuing, scrollby of the following errors. Note that this is one second's worth of log entries:

`laggy, to the point that entering pin to log in requires waiting for system to catch up between keypresses. Following log in journalctl -f results in a fast, continuing, scrollby of the following errors. Note that this is one second's worth of log entries:

Dec 01 10:06:12 danctnix kernel: hdmi-audio-codec hdmi-audio-codec.0.auto: ASoC: error at snd_soc_dai_hw_params on i2s-hifi: -19
Dec 01 10:06:12 danctnix kernel:  ff8a0000.i2s-i2s-hifi: ASoC: error at __soc_pcm_hw_params on ff8a0000.i2s-i2s-hifi: -19
Dec 01 10:06:12 danctnix kernel: hdmi-audio-codec hdmi-audio-codec.0.auto: ASoC: error at snd_soc_dai_hw_params on i2s-hifi: -19
Dec 01 10:06:12 danctnix kernel:  ff8a0000.i2s-i2s-hifi: ASoC: error at __soc_pcm_hw_params on ff8a0000.i2s-i2s-hifi: -19
Dec 01 10:06:12 danctnix kernel: hdmi-audio-codec hdmi-audio-codec.0.auto: ASoC: error at snd_soc_dai_hw_params on i2s-hifi: -19
Dec 01 10:06:12 danctnix kernel:  ff8a0000.i2s-i2s-hifi: ASoC: error at __soc_pcm_hw_params on ff8a0000.i2s-i2s-hifi: -19
Dec 01 10:06:12 danctnix kernel: hdmi-audio-codec hdmi-audio-codec.0.auto: ASoC: error at snd_soc_dai_hw_params on i2s-hifi: -19
Dec 01 10:06:12 danctnix kernel:  ff8a0000.i2s-i2s-hifi: ASoC: error at __soc_pcm_hw_params on ff8a0000.i2s-i2s-hifi: -19
Dec 01 10:06:12 danctnix kernel: hdmi-audio-codec hdmi-audio-codec.0.auto: ASoC: error at snd_soc_dai_hw_params on i2s-hifi: -19
Dec 01 10:06:12 danctnix kernel:  ff8a0000.i2s-i2s-hifi: ASoC: error at __soc_pcm_hw_params on ff8a0000.i2s-i2s-hifi: -19
Dec 01 10:06:12 danctnix kernel: hdmi-audio-codec hdmi-audio-codec.0.auto: ASoC: error at snd_soc_dai_hw_params on i2s-hifi: -19
Dec 01 10:06:12 danctnix kernel:  ff8a0000.i2s-i2s-hifi: ASoC: error at __soc_pcm_hw_params on ff8a0000.i2s-i2s-hifi: -19
Dec 01 10:06:12 danctnix kernel: hdmi-audio-codec hdmi-audio-codec.0.auto: ASoC: error at snd_soc_dai_hw_params on i2s-hifi: -19
Dec 01 10:06:12 danctnix kernel:  ff8a0000.i2s-i2s-hifi: ASoC: error at __soc_pcm_hw_params on ff8a0000.i2s-i2s-hifi: -19
Dec 01 10:06:12 danctnix kernel: hdmi-audio-codec hdmi-audio-codec.0.auto: ASoC: error at snd_soc_dai_hw_params on i2s-hifi: -19
Dec 01 10:06:12 danctnix kernel:  ff8a0000.i2s-i2s-hifi: ASoC: error at __soc_pcm_hw_params on ff8a0000.i2s-i2s-hifi: -19
Dec 01 10:06:12 danctnix kernel: hdmi-audio-codec hdmi-audio-codec.0.auto: ASoC: error at snd_soc_dai_hw_params on i2s-hifi: -19
Dec 01 10:06:12 danctnix kernel:  ff8a0000.i2s-i2s-hifi: ASoC: error at __soc_pcm_hw_params on ff8a0000.i2s-i2s-hifi: -19
Dec 01 10:06:12 danctnix kernel: hdmi-audio-codec hdmi-audio-codec.0.auto: ASoC: error at snd_soc_dai_hw_params on i2s-hifi: -19
Dec 01 10:06:12 danctnix kernel:  ff8a0000.i2s-i2s-hifi: ASoC: error at __soc_pcm_hw_params on ff8a0000.i2s-i2s-hifi: -19
Dec 01 10:06:12 danctnix kernel: hdmi-audio-codec hdmi-audio-codec.0.auto: ASoC: error at snd_soc_dai_hw_params on i2s-hifi: -19
Dec 01 10:06:12 danctnix kernel:  ff8a0000.i2s-i2s-hifi: ASoC: error at __soc_pcm_hw_params on ff8a0000.i2s-i2s-hifi: -19
Dec 01 10:06:12 danctnix kernel: hdmi-audio-codec hdmi-audio-codec.0.auto: ASoC: error at snd_soc_dai_hw_params on i2s-hifi: -19
Dec 01 10:06:12 danctnix kernel:  ff8a0000.i2s-i2s-hifi: ASoC: error at __soc_pcm_hw_params on ff8a0000.i2s-i2s-hifi: -19
Dec 01 10:06:12 danctnix kernel: hdmi-audio-codec hdmi-audio-codec.0.auto: ASoC: error at snd_soc_dai_hw_params on i2s-hifi: -19
Dec 01 10:06:12 danctnix kernel:  ff8a0000.i2s-i2s-hifi: ASoC: error at __soc_pcm_hw_params on ff8a0000.i2s-i2s-hifi: -19
Dec 01 10:06:12 danctnix kernel: hdmi-audio-codec hdmi-audio-codec.0.auto: ASoC: error at snd_soc_dai_hw_params on i2s-hifi: -19
Dec 01 10:06:12 danctnix kernel:  ff8a0000.i2s-i2s-hifi: ASoC: error at __soc_pcm_hw_params on ff8a0000.i2s-i2s-hifi: -19
Dec 01 10:06:12 danctnix kernel: hdmi-audio-codec hdmi-audio-codec.0.auto: ASoC: error at snd_soc_dai_hw_params on i2s-hifi: -19
Dec 01 10:06:12 danctnix kernel:  ff8a0000.i2s-i2s-hifi: ASoC: error at __soc_pcm_hw_params on ff8a0000.i2s-i2s-hifi: -19
Dec 01 10:06:12 danctnix kernel: hdmi-audio-codec hdmi-audio-codec.0.auto: ASoC: error at snd_soc_dai_hw_params on i2s-hifi: -19
Dec 01 10:06:12 danctnix kernel:  ff8a0000.i2s-i2s-hifi: ASoC: error at __soc_pcm_hw_params on ff8a0000.i2s-i2s-hifi: -19
Dec 01 10:06:12 danctnix kernel: hdmi-audio-codec hdmi-audio-codec.0.auto: ASoC: error at snd_soc_dai_hw_params on i2s-hifi: -19
Dec 01 10:06:12 danctnix kernel:  ff8a0000.i2s-i2s-hifi: ASoC: error at __soc_pcm_hw_params on ff8a0000.i2s-i2s-hifi: -19
[... cut ...]
Dec 01 10:06:12 danctnix kernel: hdmi-audio-codec hdmi-audio-codec.0.auto: ASoC: error at snd_soc_dai_hw_params on i2s-hifi: -19
Dec 01 10:06:12 danctnix kernel: hdmi-audio-codec hdmi-audio-codec.0.auto: ASoC: error at snd_soc_dai_digital_mute on i2s-hifi: -19
Dec 01 10:06:12 danctnix kernel:  ff8a0000.i2s-i2s-hifi: ASoC: error at __soc_pcm_hw_params on ff8a0000.i2s-i2s-hifi: -19
Dec 01 10:06:12 danctnix kernel: hdmi-audio-codec hdmi-audio-codec.0.auto: ASoC: error at snd_soc_dai_digital_mute on i2s-hifi: -19`

maxrdz, 2024-12-07:
I have a similar issue with keypresses @pbvzt3. I can send a video as well, since it is hard to put to words, but basically I can't type fast as its almost like "sticky" input.

For example, I type at a regular speed a passcode of "1234", and the actual input that went through was only 1, 2, and 4. 3 being skipped. Another example of my issue is like dragging a dropdown, such as the top bar in Phosh. It sometimes gets stuck halfway through the screen, because the touchscreen lost my finger halfway through.

maxrdz, 2024-12-11:
This issue is causing more crashes than usual on my device. Is there any good resources for me to learn debugging, to resolve this issue, or workaround it? I would greatly appreciate the help. Thanks.

maxrdz, 2024-12-11:
Errno -19 being ENODEV (no such device), I ran `lspci`.

Also, `lspci` returns the following error:

```
pcilib: Cannot open /proc/bus/pci
lspci: Cannot find any working access method.
```

Thought this was interesting.

Danct12, 2024-12-16:
Should be fixed in kernel 6.12 (a4747fbb22df8a1fd1ccd7ea928cae68e64edc94, 5c6cb98127ddb0f503c6ba0afc973194248c7c9d)

closed 2024-12-19 by Danct12

pbvzt3, 2024-12-20:
A bit late since this is already closed as complete, but was not able to test sooner. Confirming this is resolved with kernel update. Many thanks.