dreemurrs-embedded/Pine64-Arch #685: Bluetooth unable to see devices

pecosk, 2024-12-23:
**Device info (please complete the following information)**
- Device: PinePhone
- Kernel Version : 6.12.1-2-megi-danctnix
- UI: sxmo

**Describe the bug**
After recent update, bluetooth is not able to see any devices (bluetoothctl `scan on` does nothing, i've tried also GUI based tools to configure bluetooth, without success). Bluetooth is switched on, i did not see any relevant errors, just searching does nothing. 

after investigating, i've noticed that rtl8723bt-firmware package was moved to dummy and now the drivers such as rtl8723cs_xx_config.bin are now used directly from upstream linux-firmware. After searching for the original source, i've found that it was used from megous.com and tried replacing the contents of /usr/lib/firmware/rtl_bt with their version. This worked and now i can see and connect bluetooth devices.
I am not sure which files are really relevant, why it does not work using current linux-firmware package (20241210.b00a7f7e-1) and if this is the right place to report it, or i should go somewhere else so that other people don't have the same issue and it is fixed in proper way.
thx for your work

**To Reproduce**
Steps to reproduce the behavior:
1. in any kind of terminal open bluetoothctl
2. scan on
3. .. nothing happens

**Expected behavior**
a list of nearby devices should appear

Danct12, 2025-01-01:
I've been trying to get in contact with megous about this, but he is unreachable during the holidays.

jakub-bidzan, 2025-01-01:
I can confirm having the same issue on my pinephone and a friend of mine on his too. Nothing appears in device list or btctl

anarsoul, 2025-01-09:
You need https://gitlab.com/kernel-firmware/linux-firmware/-/merge_requests/398 to fix 8723cs bluetooth

pecosk, 2025-01-09:
that sounds promising, i'll check when it gets to archlinuxarm package repo, currently there is still the version from Dec 10th

Danct12, 2025-01-21:
This should now be fixed, upgrade your `linux-firmware` package.

closed 2025-01-21 by Danct12