dreemurrs-embedded/Pine64-Arch #686: phosh lock screen breaks after rotating the screen at least once

k8ieone, 2024-12-25:
**Device info (please complete the following information)**
- Device: PineTab2
- Kernel Version : 6.12.3-danctnix1-1-pinetab2
- UI: Phosh

**Describe the bug**
The lock screen issues described in #680 persist even after setting GSK_RENDERER to gl.

The phosh lockscreen doesn't show up if the device has been rotated at least once. The device still responds over SSH, the screen just stays blank.

**To Reproduce**
Steps to reproduce the behavior:
1. Install phosh
2. Lock the screen
3. Confirm the lock screen works
4. Unlock the device and rotate the screen
5. Lock the screen again and observe as the lock screen never presents

**Expected behavior**
The lock screen should work every time.

**Before submitting a bug report, please be sure that you have done the following:**
- [x] I have checked for updates (pacman -Syu) and installed the updates but the problem persisted.
- [x] I am **not** using Megi's multiboot image, which is unsupported by the developer.

Danct12, 2024-12-25:
Please attach journalctl to your bug report.

k8ieone, 2024-12-25:
```
Dec 25 14:53:55 Stan systemd[3419]: vte-spawn-e511a9ca-fe73-47ee-8afe-20890f7ac9a8.scope: Consumed 2.477s CPU time, 106.5M memory peak.
Dec 25 14:53:56 Stan systemd[3419]: dbus-:1.2-org.gnome.Console@0.service: Consumed 7.644s CPU time, 85.8M memory peak.
** FIRST ATTEMPT - SCREEN LOCK SHOWS **
Dec 25 14:53:58 Stan systemd-logind[3309]: Power key pressed short.
Dec 25 14:53:58 Stan feedbackd[3603]: Failed to set led pattern: Failed to open /sys/devices/platform/led-0/leds/white:flash/pattern: Permission denied
Dec 25 14:53:58 Stan feedbackd[3603]: Failed to set led pattern: Failed to open /sys/devices/platform/led-0/leds/white:flash/pattern: Permission denied
** MORE ATTEMPTS - ROTATING THE SCREEN VARIOUS WAYS, STILL WORKS **
Dec 25 14:54:01 Stan systemd-logind[3309]: Power key pressed short.
Dec 25 14:54:07 Stan phosh[3563]: Failed to load background image: Error opening file /usr/share/backgrounds/gnome/adwaita-d.jxl: No such file or directory
Dec 25 14:54:09 Stan systemd-logind[3309]: Power key pressed short.
Dec 25 14:54:12 Stan systemd-logind[3309]: Power key pressed short.
Dec 25 14:54:18 Stan phosh[3563]: Failed to load background image: Error opening file /usr/share/backgrounds/gnome/adwaita-d.jxl: No such file or directory
Dec 25 14:54:20 Stan phosh[3563]: Failed to load background image: Error opening file /usr/share/backgrounds/gnome/adwaita-d.jxl: No such file or directory
Dec 25 14:54:23 Stan systemd-logind[3309]: Power key pressed short.
Dec 25 14:54:23 Stan syncthing[3429]: [KRXVT] INFO: Joined relay relay://192.52.34.94:22067
Dec 25 14:54:25 Stan systemd-logind[3309]: Power key pressed short.
Dec 25 14:54:29 Stan phosh[3563]: Failed to load background image: Error opening file /usr/share/backgrounds/gnome/adwaita-d.jxl: No such file or directory
Dec 25 14:54:31 Stan phosh[3563]: Failed to load background image: Error opening file /usr/share/backgrounds/gnome/adwaita-d.jxl: No such file or directory
** LAST ATTEMPT - THIS IS WHERE THE SCREEN STAYED BLANK **
** I TURNED THE SCREEN ON AND OFF MULTIPLE TIMES, BUT THE LOCK SCREEN REFUSES TO SHOW UP **
Dec 25 14:54:31 Stan systemd-logind[3309]: Power key pressed short.
Dec 25 14:54:32 Stan systemd-logind[3309]: Power key pressed short.
Dec 25 14:54:32 Stan kernel: bes2600_wlan mmc2:0001:1: bes2600_pwr_enter_lp_mode, wait pm ind timeout
Dec 25 14:54:40 Stan kernel: bes2600_wlan mmc2:0001:1: bes2600_pwr_enter_lp_mode, wait pm ind timeout
Dec 25 14:54:47 Stan kernel: bes2600_wlan mmc2:0001:1: bes2600_pwr_enter_lp_mode, wait pm ind timeout
Dec 25 14:54:47 Stan systemd-logind[3309]: Power key pressed short.
Dec 25 14:54:49 Stan systemd-logind[3309]: Power key pressed short.
Dec 25 14:54:51 Stan systemd-logind[3309]: Power key pressed short.
Dec 25 14:54:52 Stan systemd-logind[3309]: Power key pressed short.
Dec 25 14:54:56 Stan kernel: bes2600_wlan mmc2:0001:1: bes2600_pwr_enter_lp_mode, wait pm ind timeout
Dec 25 14:55:03 Stan kernel: bes2600_wlan mmc2:0001:1: bes2600_pwr_enter_lp_mode, wait pm ind timeout
Dec 25 14:55:10 Stan kernel: bes2600_wlan mmc2:0001:1: bes2600_pwr_enter_lp_mode, wait pm ind timeout
```

k8ieone, 2024-12-25:
Added, I also added annotations to explain what was going on with the device. The weird thing is that this doesn't happen consistently.

The only suspicious message that showed up when the lock screen failed is `bes2600_wlan mmc2:0001:1: bes2600_pwr_enter_lp_mode, wait pm ind timeout` but I don't know if that could be related.

Danct12, 2024-12-25:
BES2600 errors shouldn't be related. Can you switch to another tty?

k8ieone, 2024-12-25:
Switching to a TTY doesn't work after the lock screen fail.

k8ieone, 2025-01-12:
I thought the issue was gone after the last kernel + phosh update because it hasn't happened for a really long time, but I just ran into it again today.

k8ieone, 2025-03-27:
Do you know what could be causing this? It's really annoying to be unable to use the power button to lock the screen.

k8ieone, 2025-08-12:
I have tested with both GNOME+GDM and phosh (without DM) on a fresh image.

Enabled `G_MESSAGES_DEBUG=all` to try if I get any GTK errors - nothing suspicious shows up.

This has to be something pretty low-level, GTK has no idea things are going sideways. I'm even able to unlock the screen using the clip-on keyboard after this happens (I can see the session has been unlocked through SSH). I'm guessing it's not the GPU driver, that would probably cause GTK to freak out. Maybe the screen driver?

Danct12, 2025-08-17:
@rat-h @k8ieone Can you try 6.15.10-danctnix1 kernel from the testing repo and tell me if the issue still persists?

rat-h, 2025-08-17:
@Danct12 could you please give me instruction how to try tasting repo on arch. I'm not new on Linux, but learning how to use arch ecosystem.

Danct12, 2025-08-17:
Open `/etc/pacman.conf` in a text editor (as root), scroll down to a section calls itself `[danctnix-testing], uncomment it and save.

Proceed with a normal system upgrade and it should be safe to comment it once you're done with it.

Danct12, 2025-08-17:
Now that `linux-firmware` is split, it should be safe to remove `linux-firmware-nvidia` and the other firmware packages that you're sure you didn't need.

rat-h, 2025-08-19:
I tested it a little bit more and if keyboard attached the problem is still there