dreemurrs-embedded/Pine64-Arch #687: HDMI

k8ieone, 2024-12-25:
**Device info (please complete the following information)**
- Device: PineTab2
- Kernel Version : 6.12.3-danctnix1-1-pinetab2
- UI: Phosh

**Describe the bug**
Connecting the tablet to a TV or a screen with speakers using the micro HDMI port doesn't switch the sound to HDMI. Looking through `pavucontrol` there doesn't seem to be any way to make it work. No different profile or separate device.

I also tried to turn the device off, plug in the HDMI port and only turn it on with the screen connected. Doesn't seem to change anything.

**To Reproduce**
Steps to reproduce the behavior:
1. (optional - power up the device with HDMI connected)
2. Check sound devices in settings

**Expected behavior**
Either a separate audio device or an alternative sound profile should be available when an external HDMI screen is connected.

**Before submitting a bug report, please be sure that you have done the following:**
- [x] I have checked for updates (pacman -Syu) and installed the updates but the problem persisted.
- [x] I am **not** using Megi's multiboot image, which is unsupported by the developer.

Danct12, 2025-01-02:
This should now be fixed with 6.12.7-danctnix1

closed 2025-01-02 by Danct12

k8ieone, 2025-01-06:
I just had a chance to test this today, there is a new output device called "Built-in Audio Stereo". After switching to this device, sound is being output from the connected TV/screen. Thank you :heart: 

I haven't tested if surround audio works, I assume it doesn't. Stereo is enough for my use case.