dreemurrs-embedded/Pine64-Arch #689: 2024/12/23 release, PinePhone, SXMO, conky 1.21

dukla2000, 2025-01-06:
**Device info (please complete the following information)**
- Device: PinePhone (probably also PinePhone Pro & others?)
- Kernel Version : 6.12.1-2-megi-danctnix
- UI: SXMO
- Conky version 1.21.5

**Describe the bug**
conky doesn't work - specifically no clock, date etc on home screen.

**To Reproduce**
1 Install the image https://github.com/dreemurrs-embedded/Pine64-Arch/releases/download/20241223/archlinux-pinephone-sxmo-20241223.img.xz
2 Boot the phone
3 No clock on wallpaper

**Expected behavior**
~/.config/sxmo/conky.conf should be effected

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Logfiles and additional information**
I believe well known issue - is there any way we can get back to conky 1.18
Sadly https://alaa.ad24.cz/ has vanished

**Before submitting a bug report, please be sure that you have done the following:**
- [ Yes] I have checked for updates (pacman -Syu) and installed the updates but the problem persisted.
- [ Yes] I am **not** using Megi's multiboot image, which is unsupported by the developer.

ArenM, 2025-07-23:
I can't reproduce this, I suspect it's been fixed upstream?

Sorry I haven't been very active with sxmo maintenance, I guess sometimes procrastinating can occasionally be an effective way to fix bugs though :P

dukla2000, 2025-07-24:
Happy to close this.
Still a problem as reported with plain 2024/12/23 release. However if that is now (2025/07/24) updated it works. Whether it is Conky upgrade to 1.22 or kernel to 6.15.6 or something else doesn't matter.

closed 2025-07-24 by dukla2000