dreemurrs-embedded/Pine64-Arch #693: Can't start plasma mobile on pinephone pro

Alexia-AT-Digitecnology, 2025-02-02:
**Device info (please complete the following information)**
- Device: PinePhone (Pro)
- Kernel Version : 6.12.8-1-megi-danctnix
- UI: Barebone

**Describe the bug**
Can't initialize plasma mobile

**To Reproduce**
Steps to reproduce the behavior:
1. Update the system and use the barebones version
2. Install plasma-mobile and sddm package
3. systemctl start sddm
4. Log into plasma mobile

**Expected behavior**
A plasma mobile session should start.

**Logfiles and additional information**
[Startup log](https://pastebin.lafamiliagodotera.org/XefinPUm)

Alexia-AT-Digitecnology, 2025-02-02:
Found more logs, maybe an issue with libgbm? https://pastebin.lafamiliagodotera.org/ZmKEtgvu

Alexia-AT-Digitecnology, 2025-02-02:
Fixed, uninstalled package mali-bifrost-g31-libgl-x11-gbm, and it worked :)

Danct12, 2025-02-02:
> mali-bifrost-g31-libgl-x11-gbm

We do not use that.

closed 2025-02-02 by Danct12