dreemurrs-embedded/Pine64-Arch #695: PinePhone several battery issues

h-n-a, 2025-02-18:
OG Pinephone: mobian edition.
Gnome version: 47
Kernel version: 6.12.8-1-megi-danctnix

I believe all the below issues started after the latest updates. (pacman -Syu was already ran before reporting this)

Problem 1:
-Phone claims that the battery is charged 100%.
-While plugged in to the Pine keyboard, it will only charge through the phone's USB-C port (will charge both the phone and the keyboard battery).

Problem 2:
-While the Phone is connected to the Pine keyboard the phone will turn off completely even when the phone's battery if the pine keyboard at 0%.
-The phone won't turn on until i connect the USB cable to the phone's USB port.

Problem 3:
-while this might not be related to the distro itself(may and may not be), the keyboard battery won't charge from it's own port.

Problem 4:
-The phone is draining battery even when it's completely off.
-It will drain the keyboards battery as well.

Danct12, 2025-02-18:
1. If the device claims 100% battery but shuts off when unplug then your battery did not make a good contact with the device. Try clean the battery contacts with alcohol.
2. Do not use the phone's USB port as it may damage your device. Charge the keyboard instead.
3. Install `ppkb-tools` from the repo and run `ppkb-i2c-charger-ctl`.
4. The keyboard will keep providing power to the device even when it's off.

h-n-a, 2025-02-18:
#2, the keyboard refuses to charge directly from it's port, even while the phone is connected. it will only charge if i connect the USB cable to the phone itself.
And now, the phone won't turn on for more than a few seconds while its out of the keyboard clamshell. its a really weird issue.
I've cleaned the battery and i know it works well with good capacity.