dreemurrs-embedded/Pine64-Arch #702: MMS not working

sorry-i-am-late, 2025-05-23:
**Device info (please complete the following information)**
- Device: PinePhone 
- Kernel Version : linux-megi-6.13.2-1
- UI: Phosh

**Describe the bug**
MMS has stopped working in chatty and it is acting as if such a thing never existed
**To Reproduce**
Steps to reproduce the behavior:
1. Within chatty, the upload image, send new MMS and pretty much every button related to sending an MMS is gone.
2. The MMS APN settings within chatty preferences are also grayed out and cannot be adjusted.

**Expected behavior**
I expected the buttons to at least exist even if they don't work from time to time.
**Screenshots**

[image]

straynobody, 2025-05-27:
Took me a few hours to figure out a workaround. I had installed a fresh image, and MMS wouldn't let me configure it, so it didn't work, and Chatty wouldn't open, I had to open it with the terminal. I ended up creating a new shortcut for Chatty and removing the old one. But, I did get MMS to configure and work. It was a long and bumpy road for me, but we'll try to avoid the holes along the way. Here are the steps that I did to fix it. I am going to assume you've already got it fully updated since an update seems to have broken Chatty and MMS. I flashed it to an SD Card so I could document what I did and create this guide. I'd recommend enabling ssh on the phone

```bash
sudo systemctl start sshd
```

Then, on your computer, ssh into the phone:

```bash
ssh alarm@[phone IP]
```

If you don't know the device's IP, in the terminal on the phone, type in

```bash
ip a
```

and look for the address under wlan0.

Once connected to the phone, install the required build tools and libraries:

```bash
sudo pacman -S git base-devel cmake glib2 glibc dbus libmm-glib libsoup3 \
    protobuf icu abseil-cpp meson ninja libphonenumber
```

Now, confirm libabsl is installed

```bash
ls /usr/lib/libabsl_*.so
```

We're not looking for anything particular, just to make sure they're there.

Next, clone mmsd and cd into it

```bash
git clone https://salsa.debian.org/DebianOnMobile-team/mmsd.git
cd mmsd
```

Before we build mmsd, we need to edit some configuration files. Use your favorite terminal editor, I prefer nano

```bash
nano meson.build
```

We're going to go down to the executable block (mmsd = executable....) and create a new line just above it. In that new line, we're going to paste this:

```meson
absl_log_check_op = cc.find_library('absl_log_internal_check_op')
absl_strings = cc.find_library('absl_strings')
absl_throw_delegate = cc.find_library('absl_throw_delegate')
absl_raw_hash_set = cc.find_library('absl_raw_hash_set')
absl_log_internal_message = cc.find_library('absl_log_internal_message')
absl_log_severity = cc.find_library('absl_log_severity')
absl_log_sink = cc.find_library('absl_log_sink')
absl_status = cc.find_library('absl_status')
```

In the executable block, its dependencies:, we're going to add this to it (or just replace the entire dependencies line):

```meson
dependencies: dependencies + [
  absl_log_check_op,
  absl_strings,
  absl_throw_delegate,
  absl_raw_hash_set,
  absl_log_internal_message,
  absl_log_severity,
  absl_log_sink,
  absl_status
],
```

Don't forget to add a couple spaces before each line, and the close bracket, and don't forget that trailing comma. Save and exit. Now, we're going to edit ./unit/meson.build

```bash
nano ./unit/meson.build
```

Find the line for test_phone_calls and comment it out, and near the bottom is a test() line for the test_phone_utils, comment that out as well. Save and exit. Now, you should still be in the mmsd directory, if not, we're going to go to it. Once there, we're going to run this, which will build, compile, and install mmsd

```bash
meson setup build
meson compile -C build
sudo ninja -C build install
```
Confirm it's installed

```bash
which mmsdtng
```

And should generate

`/usr/local/bin/mmsdtng`

Once that is all done, we're now going to get to making systemd service files and have them ran as the user. For some reason system level throws a fit. First, make the folder:

```bash
mkdir -p ~/.config/systemd/user
```

Create the service file

```bash
nano ~/.config/systemd/user/mmsd-tng.service
```

Paste this in there

```ini
[Unit]
Description=Multimedia Messaging Service Daemon (TNG)
After=network.target

[Service]
ExecStart=/usr/local/bin/mmsdtng
Restart=on-failure
RestartSec=2

[Install]
WantedBy=default.target
```

Save and exit. Now, have the device make it aware of the service file, and start it. It'll also start automatically after reboots

```bash
systemctl --user daemon-reexec
systemctl --user daemon-reload
systemctl --user enable --now mmsd-tng
```
If Chatty refuses to launch with the desktop launcher, and only opens with the terminal, or you haven't gotten that far yet, I'll drop in a guide to fix that bug as well.

Danct12, 2025-05-27:
I've been a bit busy lately. I'll look into this issue in a few hours. Sorry about that.

Danct12, 2025-07-23:
This should already be fixed in an upgrade that happened the same day of the last comment: https://github.com/dreemurrs-embedded/danctnix-packages/commit/da1c0b956f44a1e5a26b3934656d63d86507c0f5