dreemurrs-embedded/Pine64-Arch #707: Rear camera and LED feedback issues likely caused by kernel misconfiguration

G-P-L, 2025-07-18:
Device: PinePhone 1.2b (3/32 GB)
Kernel: 6.15.1-1-megi-danctnix
UI: Phosh 0.48.0

Due to what is likely a misconfigured kernel, the rear camera (used via libcamera/Snapshot) and LED feedback (feedbackd) do not function properly on the PinePhone.

The kernel configuration used for the PinePhone may be incorrect, which prevents any application relying on libcamera (e.g. Decoder, Snapshot) from accessing the rear camera - only the front-facing camera works. Also, some libcamera patch is needed. (Both cameras work fine with Megapixels 2, which does not use libcamera.)
Related issue: [Mobian/trixie pinephone rear camera is black and does not take photos: snapshot 47.0.1-5](https://gitlab.gnome.org/GNOME/snapshot/-/issues/240#note_2244678)

> As already mentioned in [#213 (closed)](https://gitlab.gnome.org/GNOME/snapshot/-/issues/213): https://gitlab.com/postmarketOS/pmaports/-/blob/master/temp/libcamera/0004-libcamera-simple-Skip-hwISP-formats-if-swISP-is-acti.patch is used in pmOS to work around this issue. So this is clearly a distro issue, please move it there.

> However, the bug appears to have been solved in a different way in Mobian: it seems that handling it in the kernel is the most sustainable option. The [Mobian wiki section on PinePhone cameras](https://wiki.debian.org/InstallingDebianOn/PINE64/PinePhone#Cameras) has [been updated](https://wiki.debian.org/InstallingDebianOn/PINE64/PinePhone?action=diff&rev2=34&rev1=33) to now state that `gnome-snapshot` is only expected to work on `linux 6.12`, not `linux 6.6`, and this is on the [Mobian staging release](https://packages.mobian.org).However, the bug appears to have been solved in a different way in Mobian: it seems that handling it in the kernel is the most sustainable option. The [Mobian wiki section on PinePhone cameras](https://wiki.debian.org/InstallingDebianOn/PINE64/PinePhone#Cameras) has [been updated](https://wiki.debian.org/InstallingDebianOn/PINE64/PinePhone?action=diff&rev2=34&rev1=33) to now state that gnome-snapshot is only expected to work on linux 6.12, not linux 6.6, and this is on the [Mobian staging release](https://packages.mobian.org/).

> Linux 6.12 is not needed; `sunxi64_6.6.56` from https://repo.mobian-project.org/pool/main/l/linux-6.6-sunxi64 solves the rear camera bug. This should arrive in Mobian/trixie on the time scale of a few weeks or months.

A similar misconfiguration may also cause feedbackd to always use the rear flash for feedback, even when this is not enabled and front-facing LEDs are preferred.
Related issue: [Feedback uses the flash instead of the front LEDs, regardless of the setting](https://gitlab.freedesktop.org/agx/feedbackd/-/issues/92#note_2982908)

> That's usually an issue with the kernel configuration where the PP's flash isn't exposed as a camera flash so feedbackd can't distinguish it from a status LED. Make sure your distro enables the PP's LED as camera flash.

ArenM, 2025-07-19:
Replied to the upstream feedbackd issue, I think it's a feedbackd bug.

Danct12, 2025-07-21:
Fixed in https://github.com/dreemurrs-embedded/danctnix-packages/commit/7df1dab60cea6380d1eb4a406e435af62fe4d3ca and https://github.com/dreemurrs-embedded/danctnix-packages/commit/f7d77661a031c7063b9f8e8da3c048d39626ba97

closed 2025-07-21 by Danct12