dreemurrs-embedded/Pine64-Arch #710: Newest plasma image corrupt/broken

YolotanGit, 2025-07-30:
- Device: Linux Laptop
- Kernel Version : 6.15.8-2-cachyos
- UI: Plasma Mobile

I am getting errors when trying to flash the current (20250728) plasma image.
Verification against the signature also fails:

❯ gpg --verify archlinux-pinephone-plasma-20250728.img.xz.sig archlinux-pinephone-plasma-20250728.img
gpg: Signature made Di 29 Jul 2025 19:53:15 CEST
gpg:                using RSA key F09A933C0FE0331E558CA4E166CAB7EAA45DD781
gpg: BAD signature from "Danct12 <danct12@disroot.org>" [unknown]

If this is not the correct way to report corrupt files on github, i would be happy to be educated :)

Danct12, 2025-07-30:
Please verify your md5sum

Danct12, 2025-07-30:
I didn't noticed you already extracted the image. You're supposed to verify the signature of the compressed image.

YolotanGit, 2025-07-30:
Good signature from "Danct12 <danct12@disroot.org>" [unknown]

ok, than i am having unrelated errors during flashing.
sorry for wasting your time.

closed 2025-07-30 by YolotanGit